henricm/ha-ferroamp #5: Minor features

danielolsson100, 2020-06-26:
Really nice work ! Thanx for sharing this!
I'll try to help you with the development of this module in the future if needed.

I found some minor features
no icon: 
Apparent power
ESO 19110078 State of Charge

Name change?: 
ESM %ID% SOH to ESM %ID% State of Health

Sensor value: Unknown
System State of Charge
System State of Health
Total rated capacity of all batteries
[image]

henricm, 2020-07-02:
Thanks! And thanks for the feedback.
Regarding the unknown sensors, I think this is because there's no values coming from Ferroamp API for these although they are described in the API docs. If so, they should probably be removed.

danielolsson100, 2020-07-03:
I think that the "unknown" sensor values is somehow related to the timing of data from the MQTT input.
because there is data in the database from time to time.
I can see unknown and correct values if I run the following T-SQL [SELECT * FROM states where entity_id like '%system_state_of%'] from from SQL Lite Web addon to HA.
(I have also your nice work with this sensor addon to the facebook group: https://www.facebook.com/groups/Ferroamp.Club.Sweden.)

argoyle, 2021-02-27:
I think all those points are resolved by #10 since I changed so the entities are restoring their state at startup

argoyle, 2021-05-20:
Closing since #10 is merged

closed 2021-05-20 by argoyle