henricm/ha-ferroamp #7: Add support for controlling battery charging

henricm, 2020-07-03:
Here are MQTT endpoints for manually controlling battery charging. These should be exposed as home assistant service or similar.

```
extapi/control/request
extapi/control/response
extapi/control/result
```

## Examples of request payloads from Ferroamp API doc

### Charging with 5000W:
```
extapi/control/request {"transId":"1",cmd":{"name":"charge","arg":"5000"}}
```

### Discharging with 12 kW:
```
extapi/control/request {"transId": "1508459760", "cmd":{"name":"discharge","arg":" 12000"}}
```

### Returning control of batteries to system

```
extapi/control/request {"transId": "989C6E5C-2CC1-11CA-A044-08002B1BB4F5", "cmd": {"name": "auto"}}
```

### Example Responses
```
{"status": "ack", "msg": "sending cmd to ESOs", "transId": “1”}
{"status": "nak", "msg": "Other transaction in progress", "transId": “128”}
{"status": "nak", "msg": "Max allowed power is 24000 W", "transId": “311”}
```

henricm, 2020-10-14:
Committed what I've done so far which exposes 3 home assistant services: charge, discharge and autocharge. Tested a few times in my setup and it seems to work.

Vajomi, 2020-10-15:
Thanks, it seems to work well. I have just one problem. To stop and prevent discharging, you should send "charge" (or discharge) with the value 0 W. In which format should I enter/present the value to send zero watt?

Vajomi, 2020-10-25:
I have som problems to start loading of the battery. In Home Assistant under ”Developer Tools” / ”Service” I  select ”ferroamp.charge” and try to set a value för charging current in the row ”The service dev tool allows you to call any available service in Home Assistant” for example 2000 W. I then get an error-message ”Failed to call service ferroamp/charge. Expected dict for dictionary value @ data[’service_data´]. Got 2000 ”.

It is working fine then I send ”ferroamp.autocharge” with no value and when I send value 0 W in the function ”ferroamp.charge” and ”ferroamp.discharge”. 

What am I doing wrong or is there a better way to send this command to EnergyHub and the battery? Is it easier to build a switch-command in configuration.yaml and how do I do that in that case?

I am very grateful for help and tips.

Vajomi, 2020-11-02:
> Committed what I've done so far which exposes 3 home assistant services: charge, discharge and autocharge. Tested a few times in my setup and it seems to work.

When I try to charge the battery, I do not get it to work with your example. This works for me 
extapi/control/request {"transId":"1", "cmd":{"name":"charge","arg":"5000"}}

henricm, 2020-11-03:
I've updated the docs now which hopefully clarifies a bit. The charge/discharge services takes one parameter, `power` which is the power in W (watts) you wish to charge/discharge the battery.

closed 2020-11-03 by henricm