henricm/ha-ferroamp #16: Standard data types and data length

danielolsson100, 2021-06-20:
Hi
I have been busy with other stuff so it took me some time to respond to @argoyle call for tests. and hope that @henricm agrees that it is time to fix this issue.
I have tested new version and it seems to work but there is some issues that was blocking me from using it and that is related to data types standard because I need more accurate numbers in my FA integration in HA

The old version of the code is now blocked after core-2021.5.5 due to missing key and not following the HA code standard so it is time to do something about it and summer time and more to off work coding :)

I have been using the data types below and want you align with the ide so I wont make a PR of is as of now (you can do it if you agree with the change, I have added a link to a updated script below. 

The data standard that I recommend is listed below
Integer
W - same as before
°C - No need for decimals to fill up the database. (2 dec before)
V - same as before
VA - same as before
Wh - same as before

Float with one decimal 
% - Its just not enough accurate  (INT before)
kWh - Logging energy usage per dag (INT before)
A - No need for decimals to fill up the database. (2 dec before)
kW - No need for decimals to fill up the database. (2 dec before)

I made a version of the sensor.py today that fixed the issues above and fixed som naming standard stuff etc. 
https://github.com/danielolsson100/ha-ferroamp/blob/danielolsson100-patch-1/custom_components/ferroamp/sensor.py

output example
[image]
[image]

henricm, 2021-06-21:
Thanks for your input.
I can agree that it could be good to reduce accuracy for temperature °C, if you think it's a good way to go.

Regarding kW I'm not sure why we (I) choose that for SSO and battery power instead of W.
I think it would be better to go the other way and use W everywhere.
Then the user can decide to present it in W, kW etc later depending where it's used.
But I think it would not be a good way to go to reduce accuracy in the power sensors.

danielolsson100, 2021-06-21:
I agree to run with 2 dec on ALL float values and that could be a custom setting in the future like the interval time.
The CT Clamps that EH uses for measurement in this matter is not more accurate than that anyway :)
I think its good enough with kW and W like it is today, no need to add complexity to the code if is not needed / requested ?

argoyle, 2021-06-21:
I'll take a look at the diff and create a PR (or a few perhaps) as soon as I have time

commit 3c0bfea9 (2021-06-22): feat: change default precisions and make them configurable

It's now possible to set the precision for battery, current, energy, power, temperature and voltage in the option flow.

Closes #16
commit 8d18e69b (2021-06-22): feat: change default precisions and make them configurable

It's now possible to set the precision for battery, current, energy, power, temperature and voltage in the option flow.

Closes #16
commit b7918eca (2021-06-22): feat: change default precisions and make them configurable

It's now possible to set the precision for battery, current, energy,
temperature and voltage in the option flow.

Closes #16
PR #18 (2021-06-24): feat: change default precisions and make them configurable
It's now possible to set the precision for battery, current, energy, power, temperature and voltage in the option flow.

Closes #16
closed 2021-06-24 by argoyle