henricm/ha-ferroamp #49: Remove integration name from entity name

henricm, 2021-08-02:
I'm thinking we want to skip the integration name as part of the entity name. Looking at other integrations, it seems like you don't usually include that. It's still clear from entity id and integration name where it comes from. If it's desirable, we could put it in the config flow - "Prefix entity names with integration name" as a checkbox or possibly a text field. But I'm not sure this is needed? The entity id still shows which integration is't part of, should you have multiple Energy Hubs configured.

[image]

So for example we should just use `Consumption Power`instead of `Ferroamp Consumption Power` as default name of the entities.

argoyle, 2021-08-02:
Agree! I'll make a PR. 🏃🏻

argoyle, 2021-08-02:
Hmmm, entity id is based on the name so removing that will actually remove it from entity id as well. For those who already have the integration installed the old entity id's will remain with `ferroamp_` prefix but if the integration is removed and added again (or for new users) the prefix will not be there. Not really a problem as I see it though. The name is still in the device-name.

argoyle, 2021-08-02:
PR created. Let's decide what to do with it. 😊

henricm, 2021-08-02:
I think we still want it as part of the id though - so `ferroamp_` should still prefix the entity id. But the name/friendly name should not contain the name. Does that make sense?

argoyle, 2021-08-02:
I'll see if I can find a way to add it just to the entity id.

henricm, 2021-08-02:
Sounds great. I think the name and the entity id is sent as separate arguments in the constructors and in the initial version this was also the way it was done so I hope it's doable. Let me know if you want me to look at it.
Also, thinking more about it I think it's really good if we keep the entity id:s as is to avoid trouble with migration. I actually managed to migrate without problems from a very old version (before you refactoring) to the latest one yesterday. Without the need to remove any entities. So that's great!

danielolsson100, 2021-08-03:
I have multiple Power sensors and I want to know from what sensor the data originated from.
I vote for prefix like XXXX Consumption Power, where the prefix could be null or a string in this matter so it's up to the user to decide to use the prefix or not.

henricm, 2021-08-07:
@danielolsson100 I understand your concern. However, I think it's also good to keep the config down to a minimum if possible. We already have a bit too much of config for my taste :-).
The entity id will still contain the integration name so you'll still have a way to separate sensors from your different power sensors. Then you can always arrange your Lovelace UI and group sensors by your different systems. And if you still want different display name you can always edit it in the Home assistant Entity editor manually.
So I'll still vote for removing the integration name from the entity display/friendly name (but keeping it in the id:s).

danielolsson100, 2021-08-07:
Well I an fine with this change if the ID's is not changed. Just go for it ;)

commit 63cec174 (2021-08-05): feat: remove integration name from sensor names

Closes #49
commit 1c8098bf (2021-08-09): feat: remove integration name from sensor names

Closes #49
commit 0e2d7d99 (2021-08-09): feat: remove integration name from sensor names

Closes #49
commit 249e83f7 (2021-08-09): feat: remove integration name from sensor names

Closes #49
closed 2021-08-09 by argoyle