henricm/ha-ferroamp #57: Clean up of the instruction to install

Letalis, 2021-08-05:
Hi,
I got several issues after applying the 2021.8 update.
And the Ferroamp integration doesn't work anymore.

And when I cleaned out the custom_component and follows your guide it doesn't match

Letalis, 2021-08-05:
I cant start listening and are getting info in the MQTT broker , But no entities are displayed in HA

Letalis, 2021-08-05:
Now it found everything.. Quite slow.. Not sure if its my environment or something else.

argoyle, 2021-08-05:
Depending on what hardware you have discovery can take up to a minute. Other than that it shouldn't be slow I would say.

argoyle, 2021-08-06:
If there are things in the documentation that is incorrect we welcome a PR with updates. 😊

argoyle, 2021-08-10:
Closing this for now since you got it running.

closed 2021-08-10 by argoyle