henricm/ha-ferroamp #59: Service calls

TurboJonte, 2021-08-05:
When making a service call to control a battery the request is being published continuously to the MQTT topic resulting in the response "Other transaction in progress". This also seems to apply to the request for checking the version of the external API, this could be done with a lower frequency, probably 60s or more.

For each request the the response message should be checked for the corresponding "transId" and if the status is "ack" or not before retrying after a given time.

argoyle, 2021-08-06:
That's strange. We actually don't retry at all. Are you talking directly to the MQTT-broker in the EnergyHub or do you have a bridged connection? Could there be some retain-configuration there?

TurboJonte, 2021-08-06:
It's probably not related to the integration. Yes, I have a bridge connection. I tried with publishing a message to the request topic directly to the EnergyHub and it keeps repeating the message. Never had this issue before, have been running a bridge connection for well over a year now without issues. I changed the "try_private" property in the bridge-config from false to true and this seemes to help, the message now only seems to repeat a couple of times and not indefinitely.

argoyle, 2021-08-06:
I use default settings for my bridge connection and have never seen anything like that but I'm far from a MQTT expert. 🤷🏼‍♂️

TurboJonte, 2021-08-06:
Never had any issues previously with the default settings. Not sure when the issue started but I've added another bridge to a CTEK charger though, but can't see why that should cause this.

argoyle, 2021-08-06:
Are you bridging all topics or just each subset for each bridge? Just thinking that if you bridge everything maybe the charger does some kind of magic?

argoyle, 2021-08-10:
I'll close this since it seems to be something in your environment rather than in the integration

closed 2021-08-10 by argoyle

TurboJonte, 2021-08-10:
> I'll close this since it seems to be something in your environment rather than in the integration

Yeah, I got it fixed.