henricm/ha-ferroamp #72: MQTT - Mosquitto broker

bopbrev, 2021-09-01:
Hi, I have installed Ferroamp, and everything seems to work so far. It has found my local IP. My challenge is to set up Configuring Bridges in my Mosquitto broker. Mosquito are already set up and serves the tasmota. I have received the name and password from ferroamp to my Energy MQTT broker. Is it possible for someone to make an example of how this can be set up and where this is done?

argoyle, 2021-09-01:
If you have Mosquitto installed as a supervised addon you need to add the following to your config:
```
customize:
  active: true
  folder: mosquitto
```

That requires a folder named mosquitto under `/share` and in there I have added a file called `energyhub.conf` with the following content (change accordingly for your setup):
```
connection energyhub
address <your energyhub ip-address>
remote_username <username from Ferroamp>
remote_password <password from Ferroamp>
topic extapi/# both
```

Restart mosquitto and make sure it says it's connecting to your EnergyHub.

bopbrev, 2021-09-01:
Hi, thanks for the suggestion. After some exploration of being able to insert the right file in the right folder, it seems to play, but I get an error message:
-------------------------------------log------------------------
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] mosquitto.sh: executing... 
[00:09:02] INFO: Certificates found: SSL is available
[cont-init.d] mosquitto.sh: exited 0.
[cont-init.d] nginx.sh: executing... 
[cont-init.d] nginx.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[00:09:03] INFO: Starting NGINX for authentication handling...
[00:09:03] INFO: Starting mosquitto MQTT broker...
1630534143: Loading config file /share/mosquitto/energyhub.conf
1630534143: mosquitto version 1.6.12 starting
1630534143: |-- *** auth-plug: startup
1630534143: Config loaded from /etc/mosquitto/mosquitto.conf.
1630534143: Loading plugin: /usr/share/mosquitto/auth-plug.so
1630534143:  ├── Username/password checking enabled.
1630534143:  ├── TLS-PSK checking enabled.
1630534143:  └── Extended authentication not enabled.
1630534143: Opening ipv4 listen socket on port 1883.
1630534143: Opening ipv6 listen socket on port 1883.
1630534143: Opening websockets listen socket on port 1884.
1630534144: Opening ipv4 listen socket on port 8883.
1630534144: Opening ipv6 listen socket on port 8883.
1630534144: Opening websockets listen socket on port 8884.
1630534144: Warning: Mosquitto should not be run as root/administrator.
1630534144: Connecting bridge energyhub (192.168.0.2:1883)
1630534144: mosquitto version 1.6.12 running
1630534144: New connection from 127.0.0.1 on port 1883.
1630534144: Socket error on client local.core-mosquitto.energyhub, disconnecting.
1630534144: Socket error on client <unknown>, disconnecting.
------------------------------------------------
Any suggestions on where to look for the error?

argoyle, 2021-09-02:
I've got lots of `1630559603: Socket error on client <unknown>, disconnecting.` in my logs as well but nothing regarding the energyhub. Do you get information from Ferroamp into the system? In that case I would probably ignore the socket errors. 😊

bopbrev, 2021-09-02:
Hi, this is what I got when I installed the solution (before your setup), and it's the same I find now. (see picture)
[image]

argoyle, 2021-09-02:
Control Status and Extapi Version are our sensors but the rest is probably from some router or something like that. Looking at the information on control status I assume that the MQTT-configuration you have is to the mosquitto-server (rather than directly to ENergyHub) and that the bridge is not working since you would have quite a lot of sensors otherwise. Not sure what we can do to help you investigate though.

bopbrev, 2021-09-02:
Now see that those sensors come from the UniFi router. 
Should the sensors appeared on the list in Mosquitto broker integration if everything was working correct ?
[image]

argoyle, 2021-09-02:
No, they should be under the Ferroamp integration. Did you use HACS to install the integration? I would recommend to remove your current configuration of the integration and re-add it and see if that makes any difference. If it doesn't I guess the bridge is not working correctly and no actual information is making it's way into mosquitto.

argoyle, 2021-09-02:
This is how it looks in my integration-page
[image]

bopbrev, 2021-09-02:
OK, I did uninstall, and tryed to install a new one, but getting following message:
[image]

argoyle, 2021-09-02:
Correct. The latest version requires the latest version of Home Assistant. Either upgrade your installation or choose version 1.2.1 of Ferroamp in the drop down.

bopbrev, 2021-09-03:
Hi, then I have updated and reinstalled ferroamp, but unfortunately there are no more sensors. Attach some pictures of my setup in case you can see that there is something here I have misunderstood
[image]
[image]
[image]

argoyle, 2021-09-03:
Everything looks correct to me. You run the same version of mosquitto as I do and everything seems to be correctly setup. What happens if you subscribe to `extapi/#` if you click configure on the MQTT integration? If the bridge is working you should get one message per second from the EnergyHub. My feeling is that there is something acting up in your network. Could you install a MQTT client and connect it directly to the EnergyHub with the same config as in the bridge-config perhaps.

TurboJonte, 2021-09-06:
I can recommend [MQTT Explorer](http://mqtt-explorer.com/) for testing of your Mosquitto  configuration.

bopbrev, 2021-09-07:
Thanks, I see that I get data from the test setup for mosquitto (pictures 1 and 2) 
[image]
[image]
and from the setup for my home assistant (pictures 3 and 4), 
[image]
[image]
but get an error message with connection to ferroamp (picture 5)
[image]

bopbrev, 2021-09-07:
Could there be any missing setup in the local Ferroamp?

argoyle, 2021-09-07:
So either Ferroamp didn't actually enable the MQTT-broker (if there is actually anything for them to enable, could be that it's always enabled, not sure about that) or your network is acting up somehow. I know that the ip-address you used is the same as you had in the router-sensor but try to make sure that that is actually the adress of the Energy hub.

bopbrev, 2021-09-10:
Hi and thanks for the very good follow-up :), MQTT had to be turned on in the local ferroamp from the supplier. Everything now seems to work.

closed 2021-09-10 by bopbrev

argoyle, 2021-09-10:
Nice to hear. 😊