henricm/ha-ferroamp #76: sensor.ferroamp_total_solar_energy suddenly decreasing in value

jonasbkarlsson, 2021-09-14:
The new Energy feature of HA is great, but it heavily depends on the inputs to be "state_class":"total_increasing".

I had a sudden drop in the `sensor.ferroamp_total_solar_energy` (from 532.2 to 443.5), which messed up the Energy feature. 

From the database, `select * from states where entity_id="sensor.ferroamp_total_solar_energy";`

```
6724537|sensor|sensor.ferroamp_total_solar_energy|532.2|{"state_class":"total_increasing","unit_of_measurement":"kWh","friendly_name":"Total Solar Energy","icon":"mdi:solar-power","device_class":"energy"}|7436277|2021-09-14 07:10:12.056101|2021-09-14 07:10:12.056101|2021-09-14 07:10:12.056101|6724087
6724760|sensor|sensor.ferroamp_total_solar_energy|443.5|{"state_class":"total_increasing","unit_of_measurement":"kWh","friendly_name":"Total Solar Energy","icon":"mdi:solar-power","device_class":"energy"}|7436523|2021-09-14 07:13:14.061745|2021-09-14 07:13:14.061745|2021-09-14 07:13:14.061745|6724537
6724796|sensor|sensor.ferroamp_total_solar_energy|532.2|{"state_class":"total_increasing","unit_of_measurement":"kWh","friendly_name":"Total Solar Energy","icon":"mdi:solar-power","device_class":"energy"}|7436563|2021-09-14 07:13:45.058075|2021-09-14 07:13:45.058075|2021-09-14 07:13:45.058075|6724760
```
What could be the cause of this?

I have 2 SSOs, and both of the `sensor.ferroamp_sso_XXX_total_energy` look ok. At this time they had the values 307.3 kWh and 224.9 kWh. So none of them were 443.5 kWh

I have manually fixed the statistics table to fix the Energy view. So I'm not in a urgent need for help, but it would be nice to avoid this in the future.

argoyle, 2021-09-14:
That sounds like a question to Ferroamp I would say.

jonasbkarlsson, 2021-09-14:
Right, the source of the problem could be Ferroamp. BTW, I have had similar issues with data from the Nibe component. Sure it would be nice if all sources of data provides error-free information, but considering all the issues I have had the last month, I doubt that is a reasonable expectation. So my conclusion is rather that HA as a whole needs to be more robust. And maybe it the Energy feature that needs improvements, instead of each individual component.

argoyle, 2021-09-14:
The "contract" of state class total increasing is that the value always increases. If the value decreases for some reason, the sum is reset to zero and then start increasing from there again which would give you a spike in usage. I had the same problem a week ago but my spike was like 10000kWh 😳 making my graphs very difficult to read.

argoyle, 2021-09-14:
I think they are introducing a new state class in the next release which might be more suitable that will cope with those kinds of changes. I'll leave the issue open until I have investigated.

commit d389ca10 (2021-09-17): feat: make sure state is always increasing for energy sensors

Closes #76
closed 2021-09-27 by argoyle