henricm/ha-ferroamp #85: 1.4.0 fix/update not working.

u1malmberg, 2021-11-04:
Just upgraded from 1.2 to 1.4, the fix for #76 is breaking the upgrade.  For some reason most of my aggregating entities (like total_solar_energy etc) fail to be loaded,  they have the value "unknown" after upgrade.  Looks ok in the data stream, so must be lost in the conversion somehow, but this causes the new code to break:
```
val = round(temp / count / 3600000000, self._precision)
            if self._attr_native_value is None or self._attr_state_class != STATE_CLASS_TOTAL_INCREASING or val > float(self._attr_native_value):
                self._attr_native_value = val
                if self._precision == 0:
                    self._attr_native_value = int(self._attr_native_value)
                return True
            else:
                return False
```
as the "float(self._attr_native_value)" fails when not able to convert "unknown" to a float.

I am having fairly frequent issues with the issue this fix is suppose to fix, unfortunately my python skills are lacking to more than for troubleshooting. :)  I reverted the if statement to get it working and now values are showing correctly again.

results after upgrading:
[image]

* Exception in ehub_event_received when handling msg on 'extapi/data/ehub': '{"wloadconsq": {"L2": "11798853816863", "L3": "10979673467307", "L1": "6638586388368"}, "iloadd": {"L2": "2.53", "L3": "1.55", "L1": "0.45"}, "wextconsq": {"L2": "7946365464096", "L3": "6802679818174", "L1": "4436621595469"}, "ppv": {"val": "0.00"}, "iext": {"L2": "3.40", "L3": "2.37", "L1": "1.16"}, "iloadq": {"L2": "3.07", "L3": "2.01", "L1": "0.75"}, "iace": {"L2": "0.00", "L3": "0.00", "L1": "0.00"}, "ul": {"L2": "235.77", "L3": "235.20", "L1": "235.82"}, "pinvreactive": {"L2": "138.37", "L3": "138.04", "L1": "140.07"}, "ts": {"val": "2021-11-03T01:00:16UTC"}, "ploadreactive": {"L2": "421.79", "L3": "257.78", "L1": "75.04"}, "state": {"val": "4101"}, "wloadprodq": {"L2": "19096447874", "L3": "17230314640", "L1": "20651676996"}, "iavbl": {"L2": "9997.77", "L3": "9998.52", "L1": "9999.42"}, "pinv": {"L2": "13.34", "L3": "13.30", "L1": "11.67"}, "iextq": {"L2": "3.15", "L3": "2.09", "L1": "0.82"}, "pext": {"L2": "525.15", "L3": "347.59", "L1": "136.73"}, "wbatcons": {"val": "4392639457542"}, "wextprodq": {"L2": "4370649133629", "L3": "3826985300093", "L1": "5595077240829"}, "wpv": {"val": "42994805534500"}, "winvconsq": {"L2": "95153962670", "L3": "116709288236", "L1": "108896629155"}, "pextreactive": {"L2": "560.16", "L3": "395.82", "L1": "215.11"}, "udc": {"neg": "-375.42", "pos": "375.79"}, "sext": {"val": "1632.59"}, "pbat": {"val": "0.00"}, "iextd": {"L2": "3.36", "L3": "2.38", "L1": "1.29"}, "iavblq_3p": {"val": "9996.85"}, "wbatprod": {"val": "5241033780070"}, "iavblq": {"L2": "9996.85", "L3": "9997.91", "L1": "9999.18"}, "ild": {"L2": "0.83", "L3": "0.83", "L1": "0.84"}, "pload": {"L2": "511.81", "L3": "334.29", "L1": "125.06"}, "ilq": {"L2": "0.08", "L3": "0.08", "L1": "0.07"}, "winvprodq": {"L2": "8079790981513", "L3": "7900565770117", "L1": "7761650195025"}, "il": {"L2": "0.62", "L3": "0.62", "L1": "0.63"}}' Traceback (most recent call last): File "/config/custom_components/ferroamp/sensor.py", line 140, in ehub_event_received update_sensor_from_event(event, ehub, store) File "/config/custom_components/ferroamp/sensor.py", line 134, in update_sensor_from_event sensor.add_event(event) File "/config/custom_components/ferroamp/sensor.py", line 626, in add_event self.process_events(now) File "/config/custom_components/ferroamp/sensor.py", line 633, in process_events if self.update_state_from_events(temp): File "/config/custom_components/ferroamp/sensor.py", line 856, in update_state_from_events or val > float(self._attr_native_value): ValueError: could not convert string to float: 'unknown'

argoyle, 2021-11-04:
Hmmm, that is strange. Will take a closer look tonight.

argoyle, 2021-11-04:
I assume that you run at least Home Assistant 2021.9?

That last exception is using your own version, right? The line numbers do not correspond to the 1.4.0 version.

u1malmberg, 2021-11-04:
Running core-2021.10.7
Might be from a later run where I tried to fix it... but it was pointing to that particular line...  But ~10 lines off does not really make sense...  I still have the original file (attached here) and the the row numbers does not really match there either with the commit for the #76 fix! :)   Not sure how I can have messed this up, updated via HACS when it prompted me for an update.  But then again if the sensor.py file was not updated the "always increasing" fix would not be there.
[sensor copy.txt](https://github.com/henricm/ha-ferroamp/files/7477449/sensor.copy.txt)
.

argoyle, 2021-11-04:
Hmmm, so the problem is actually in the if-statement where it checks that the new value is greater than the current value and your current value is not a value that is convertable to a float. I'll write a test for that and then add a fix for it.

argoyle, 2021-11-05:
I have release v1.4.1 which should solve this. Please update and test @u1malmberg

closed 2021-11-05 by argoyle

u1malmberg, 2021-11-05:
@argoyle  looks like it is working, I do not get that same issue any more.  Thanks!! 🙏 
I did get my daily utility_meter measurement messed up...again!  hopefully for the last time! :)  Was getting tired of correcting the data in influx directly, so looking forward to having that fix in place! 
[image]

argoyle, 2021-11-05:
Thanks for finding and reporting. Hopefully the last time your utility meter did that. 😄

commit b0272878 (2021-11-04): fix: handle unknown value for energy sensor

Fixes #85
PR #86 (2021-11-05): fix: handle unknown value for energy sensor
Fixes #85