henricm/ha-ferroamp #92: No history for SSO Relay Status in history-graph

bj00rn, 2021-11-21:
Thanks for a great integration, works like a charm!

I would like to show a history-graphs ideally a bars for various state sensors but i can't get it to work properly.

I've been mucking around with SSO Relay Status with the configuration below.

Seems there is no state history for this sensor, is this expected behaviour?

Am I missing something, im pretty new to HA.

```yaml
type: history-graph
entities:
  - entity: sensor.energyhub_14_sso_21020540_relay_status
  - entity: sensor.energyhub_14_sso_21020540_relay_status
  - entity: sensor.energyhub_14_sso_21030452_relay_status
  - entity: sensor.energyhub_14_sso_21030613_relay_status
hours_to_show: 24000
refresh_interval: 1
```

[image]

Thanks again!

argoyle, 2021-11-21:
Nice find! I think I've actually seen it but not thought much about it. 😊 Will release the fix as soon as the build is successful. Thanks for reporting.

closed 2021-11-21 by argoyle

bj00rn, 2021-11-21:
> Nice find! I think I've actually seen it but not thought much about it. 😊 Will release the fix as soon as the build is successful. Thanks for reporting.

Great! I think the same goes for most state sensors.

Preferably the state sensors should report string values right? That way HA defaults to `bar-style` charts. Might be problematic if state codes are unknown though..

argoyle, 2021-11-21:
That could be true. I'll take a look at the other ones tomorrow then. 😄 

Your existing history will still be wrong btw so you might have to wait for the new version to have been running for 24h before it looks nice.

argoyle, 2021-11-22:
I think all state sensors are covered by this change (at least the ones I have). If you find any that I have missed, please open another issue.

commit 6ceb2f07 (2021-11-21): fix: make relay status history available

The problem was that unit of measurement was set to empty string instead of None.
Started to introduce strict typing to allow it to be either a string or
None.

Closes #92
PR #93 (2021-11-21): fix: make relay status history available
The problem was that unit of measurement was set to empty string instead of None.
Started to introduce strict typing to allow it to be either a string or
None.

Closes #92