henricm/ha-ferroamp #95: Add sensor for measurement "iavbl"

jonasbkarlsson, 2021-11-27:
Measurements `iavblq_3p` and `iavblq` are currently exposed as sensors. However, CTEK Chargestorm 2 (SW ccu_R3.11.18.2) is using the measurement `iavbl `for its integration with the EnergyHub.

Also, note that  `iavblq_3p` and `iavblq` are calculated as `main fuse` minus `peak current`, which is really strange (and useless?), since `main fuse` is a RMS value (RMS value - peak value).

On the other hand, `iavbl` is calculated as `main fuse` minus `RMS current`, which makes sense (RMS value - RMS value).

So is anyone actually using `iavblq_3p` and `iavblq`?

argoyle, 2021-11-29:
Those were added in #58. I don't use them myself (I guess something needs to be configured for my EnergyHub to even export them) and is far from an expert in electrics/energy/power. Maybe @TurboJonte and @danielolsson100 can chime in here?

As long as someone can tell me how something should work (preferably with a good example which can be encoded as a unit-test) I can probably do the implementation. I would also need the JSON from the EnergyHub where the values are present.

danielolsson100, 2021-12-01:
I don't use those values for load balancing, perhaps because I am lazy :)

I use something like this to calculate available Grid Load in realtime and then remove the load from the charger.
The ACE functionality works great and handle the balance between the phases so I don't need to worry on that.
There will always be som energy over because I am converting the datatype to INT in end.  

  - platform: template
    sensors:
      available_grid_load:
        value_template: "{{(230*3*16/1000 - (states.sensor.Realtime_Grid_Power.state | float)) | int}}"
        friendly_name: "Available Grid Load"
        unit_of_measurement: "kW"

The realtime values is also used within configuration.yaml
  - platform: mqtt
    state_topic: "extapi/data/ehub"
    unique_id: "ehub/pext"
    name: "Realtime Grid Power"
    unit_of_measurement: "kW"
    value_template: "{{ ((((value_json.pext.L1) | int + (value_json.pext.L2) | int + (value_json.pext.L3) | int)) / 1000) | round(2) }}"
    
    This is displayed in the graph below
    
[image]

TurboJonte, 2021-12-04:
> From reading the source code, I think the following changes would be needed:
> 
> In sensor.py, insert the below after row 1493:
> 
> ```
> ThreePhaseMinFerroampSensor(
>   "Available RMS current for load balancing",
>   slug,
>   "iavbl",
>   ELECTRIC_CURRENT_AMPERE,
>   "mdi:current-ac",
>   f"{slug}_{EHUB}",
>   EHUB_NAME,
>   interval,
>   precision_current,
>   config_id,
> ),
> ```
> 
> I don't know if `state_class=STATE_CLASS_MEASUREMENT,` should be included or not. But maybe not since it's not included for the measurement `iavblq`.
> 
> In test_sensor.py, insert the below after row 519:
> 
> ```
>     state = hass.states.get("sensor.ferroamp_available_rms_current_for_load_balancing")
>     assert state.state == "26"
>     assert state.attributes == {
>         'L1': 26.31,
>         'L2': 29.69,
>         'L3': 31.20,
>         'device_class': 'current',
>         'friendly_name': 'Available RMS current for load balancing',
>         'icon': 'mdi:current-ac',
>         'state_class': 'measurement',
>         'unit_of_measurement': 'A'
>     }
> ```

This seems correct, I think we should add the sensor. I've made a pull request.

closed 2021-12-04 by argoyle

jonasbkarlsson, 2021-12-06:
Thanks! Looks good!

PR #98 (2021-12-04): feat: add sensor for load balancing
New sensor named "Available RMS current for load balancing"

Closes #95 

Supersedes #96 and #97