henricm/ha-ferroamp #108: Suggestion: Separate sensors for phase load

robinostlund, 2021-12-20:
Hi,
Would first of start to say what an amazing integration, got my energyhub last week. So great work!!
One suggestion from me would be to have separate sensors for each current load, as i would like to monitor my main fuses :)

What do you think about that? :)

Brg
Robin

danielolsson100, 2021-12-20:
If you add this to configuration.yaml and define ACE to 16A (or similar..) then should you be good to go to integrate it to whatever.
You don't need to worry about the phase balancing since the Adaptive Currency equalization (ACE) in the energy hub (EH) is handling that.
I use it to limit the immersion heater within my Nibe 1210-7 with a relay and a esp8266 with esphome.

```  
- platform: template
    sensors:
      grid_overload:
        friendly_name: "Grid Overload"
        #delay_off:
        #  minutes: 5
        delay_on:
          minutes: 5
        value_template: >-
          {{ states('sensor.ferroamp_grid_power') | int < -11040 }}
```
And if you add this to your automation.yaml the you can use it tp send push notice with Tibber as an example.
```
- id: '123456789'
  alias: Grid Overload
  description: ''
  trigger:
  - platform: state
    entity_id: binary_sensor.grid_overload
    from: 'off'
    to: 'on'
  condition: []
  action:
  - service: notify.tibber
    data:
      message: Grid - Overload
      title: Grid Overload
  mode: single
  ```

danielolsson100, 2021-12-20:
You can for sure add the L1,L2,L3 as individual sensors if you feel for it as a custom sensor with the template functionality

```
{{state_attr('sensor.ferroamp_grid_current','L1')}}
{{state_attr('sensor.ferroamp_grid_current','L2')}}
{{state_attr('sensor.ferroamp_grid_current','L3')}}
```
I use it to calculate available grid load in *near realtime* as an example below to publish it to my kitchen energy display and to feed the data to my Easee charger as a dynamic fuse load logic.
```
  - platform: mqtt
    state_topic: "extapi/data/ehub"
    unique_id: "ehub/pext"
    name: "Realtime Grid Power"
    unit_of_measurement: "kW"
    value_template: "{{ ((((value_json.pext.L1) | int + (value_json.pext.L2) | int + (value_json.pext.L3) | int)) / 1000) | round(2) }}"

  - platform: template
    sensors:
      available_grid_load:
        value_template: "{{(230*3*16/1000 - (states.sensor.Realtime_Grid_Power.state | float)) | int}}"
        friendly_name: "Available Grid Load"
        unit_of_measurement: "kW"
```

robinostlund, 2021-12-22:
Thanks, got it :) missed that grid_current had those attributes :)

closed 2021-12-22 by robinostlund