henricm/ha-ferroamp #126: sensor.ferroamp_external_energy_produced is not strictly increasing

jonasbkarlsson, 2022-02-16:
The `sensor.ferroamp_external_energy_produced` has suddenly started to oscillate between two values. It makes the Energy Dashboard in HA look weird.
[image]
Maybe this is a bug in the EnergyHub, but would it be possible for the integration to ensure that the values delivered are strictly increasing?

Maybe not so easy to achieve, without causing other problems...

argoyle, 2022-02-16:
Hmmm, thought I had fixed that one already.

argoyle, 2022-02-16:
Can you check the events from the EnergyHub via a MQTT client and see if the values are actually oscillating?

argoyle, 2022-02-16:
It's the `wextprodq` attribute you are interested in

jonasbkarlsson, 2022-02-16:
The oscillations have now stopped, since I'm now exporting to the grid. Should have checked the EnergyHub before. However, I also have Node-RED/Grafana, and I can see the same strange results there. See the blue (Exported to grid) during the last night. So it's quite clear that the data from the EnergyHub has been strange.
[image]

argoyle, 2022-02-16:
I misrembered regarding fixing the issue though, that was just for the "regular" energy-sensors, not the ThreePhase sensors. I'll make the same change there later. That will make the value strictly increasing.

argoyle, 2022-02-20:
@jonasbkarlsson Please have a look at the #129 It should solve your problem for next time.

PR #129 (2022-02-21): feat: make sure that 3-phase energy sensors are strictly increasing
Closes #126
commit 9446217e (2022-02-21): feat: make sure that 3-phase energy sensors are strictly increasing

Closes #126
closed 2022-02-21 by argoyle