henricm/ha-ferroamp #162: Frequent log error

bj00rn, 2022-04-30:
Getting the below error in log.

Logger: homeassistant.util.logging
Source: util/logging.py:114
First occurred: 14 april 2022 14:40:57 (89436 occurrences)
Last logged: 09:38:30

Exception in ehub_event_received when handling msg on 'extapi/data/ehub': '{"wloadconsq": {"L2": "22297530351747", "L3": "23128929514531", "L1": "15106482872539"}, "iloadd": {"L2": "0.120", "L3": "-0.416", "L1": "-0.061"}, "wextconsq": {"L2": "19975740691464", "L3": "20570665882684", "L1": "13471409102115"}, "ppv": {"val": "9173.195"}, "iext": {"L2": "12.596", "L3": "12.867", "L1": "12.439"}, "iloadq": {"L2": "0.173", "L3": "-0.100", "L1": "0.525"}, "iace": {"L2": "0.000", "L3": "0.000", "L1": "0.000"}, "ul": {"L2": "230.956", "L3": "231.420", "L1": "231.171"}, "pinvreactive": {"L2": "269.952", "L3": "273.113", "L1": "284.425"}, "ts": {"val": "2022-04-30T07:37:29UTC"}, "ploadreactive": {"L2": "19.597", "L3": "-68.074", "L1": "-9.971"}, "state": {"val": "4101"}, "wloadprodq": {"L2": "3677347997", "L3": "5058439197", "L1": "3783842710"}, "pinv": {"L2": "-2905.948", "L3": "-2939.932", "L1": "-2933.173"}, "iextq": {"L2": "-17.621", "L3": "-18.066", "L1": "-17.419"}, "pext": {"L2": "-2877.695", "L3": "-2956.296", "L1": "-2847.355"}, "wextprodq": {"L2": "2197000372066", "L3": "2134889525966", "L1": "2756603170887"}, "wpv": {"val": "14004333677939"}, "winvconsq": {"L2": "58709692914", "L3": "37641033192", "L1": "122371452668"}, "pextreactive": {"L2": "289.550", "L3": "205.039", "L1": "274.454"}, "udc": {"neg": "-380.526", "pos": "381.101"}, "sext": {"val": "8762.339"}, "iextd": {"L2": "1.773", "L3": "1.253", "L1": "1.679"}, "ild": {"L2": "1.653", "L3": "1.669", "L1": "1.740"}, "pload": {"L2": "28.253", "L3": "-16.364", "L1": "85.818"}, "ilq": {"L2": "-17.794", "L3": "-17.966", "L1": "-17.944"}, "winvprodq": {"L2": "4387765307717", "L3": "4530226752292", "L1": "4384214075035"}, "il": {"L2": "12.639", "L3": "12.758", "L1": "12.750"}}' Traceback (most recent call last): File "/config/custom_components/ferroamp/sensor.py", line 137, in ehub_event_received update_sensor_from_event(event, ehub, store) File "/config/custom_components/ferroamp/sensor.py", line 131, in update_sensor_from_event sensor.add_event(event) File "/config/custom_components/ferroamp/sensor.py", line 623, in add_event self.process_events(now) File "/config/custom_components/ferroamp/sensor.py", line 630, in process_events if self.update_state_from_events(temp): File "/config/custom_components/ferroamp/sensor.py", line 752, in update_state_from_events pct = int(float(self.state) / 10) * 10 ValueError: could not convert string to float: 'unknown'
Exception in ehub_event_received when handling msg on 'extapi/data/ehub': '{"wloadconsq": {"L2": "22297543917058", "L3": "23128938050085", "L1": "15106489483673"}, "iloadd": {"L2": "-1.136", "L3": "-0.414", "L1": "1.444"}, "wextconsq": {"L2": "19975740691464", "L3": "20570665882684", "L1": "13471409102115"}, "ppv": {"val": "9184.925"}, "iext": {"L2": "10.836", "L3": "12.909", "L1": "11.000"}, "iloadq": {"L2": "2.647", "L3": "-0.101", "L1": "2.847"}, "iace": {"L2": "0.000", "L3": "0.000", "L1": "0.000"}, "ul": {"L2": "230.502", "L3": "230.921", "L1": "230.577"}, "pinvreactive": {"L2": "268.118", "L3": "271.708", "L1": "282.227"}, "ts": {"val": "2022-04-30T07:37:55UTC"}, "ploadreactive": {"L2": "-185.156", "L3": "-67.600", "L1": "235.433"}, "state": {"val": "4101"}, "wloadprodq": {"L2": "3677347997", "L3": "5058604594", "L1": "3783842710"}, "pinv": {"L2": "-2910.341", "L3": "-2943.063", "L1": "-2932.483"}, "iextq": {"L2": "-15.209", "L3": "-18.125", "L1": "-15.139"}, "pext": {"L2": "-2478.908", "L3": "-2959.555", "L1": "-2468.301"}, "wextprodq": {"L2": "2197062373972", "L3": "2134957641965", "L1": "2756672866318"}, "wpv": {"val": "14004563184133"}, "winvconsq": {"L2": "58709692914", "L3": "37641033192", "L1": "122371452668"}, "pextreactive": {"L2": "82.962", "L3": "204.107", "L1": "517.660"}, "udc": {"neg": "-380.522", "pos": "381.046"}, "sext": {"val": "8015.026"}, "iextd": {"L2": "0.509", "L3": "1.250", "L1": "3.175"}, "ild": {"L2": "1.645", "L3": "1.664", "L1": "1.731"}, "pload": {"L2": "431.433", "L3": "-16.492", "L1": "464.182"}, "ilq": {"L2": "-17.856", "L3": "-18.024", "L1": "-17.986"}, "winvprodq": {"L2": "4387840874949", "L3": "4530303238448", "L1": "4384290381611"}, "il": {"L2": "12.677", "L3": "12.799", "L1": "12.782"}}' Traceback (most recent call last): File "/config/custom_components/ferroamp/sensor.py", line 137, in ehub_event_received update_sensor_from_event(event, ehub, store) File "/config/custom_components/ferroamp/sensor.py", line 131, in update_sensor_from_event sensor.add_event(event) File "/config/custom_components/ferroamp/sensor.py", line 623, in add_event self.process_events(now) File "/config/custom_components/ferroamp/sensor.py", line 630, in process_events if self.update_state_from_events(temp): File "/config/custom_components/ferroamp/sensor.py", line 752, in update_state_from_events pct = int(float(self.state) / 10) * 10 ValueError: could not convert string to float: 'unknown'
Exception in ehub_event_received when handling msg on 'extapi/data/ehub': '{"wloadconsq": {"L2": "22297545643431", "L3": "23128938050085", "L1": "15106491331476"}, "iloadd": {"L2": "-1.135", "L3": "-0.414", "L1": "1.442"}, "wextconsq": {"L2": "19975740691464", "L3": "20570665882684", "L1": "13471409102115"}, "ppv": {"val": "9185.514"}, "iext": {"L2": "10.851", "L3": "12.910", "L1": "10.988"}, "iloadq": {"L2": "2.643", "L3": "-0.101", "L1": "2.843"}, "iace": {"L2": "0.000", "L3": "0.000", "L1": "0.000"}, "ul": {"L2": "230.920", "L3": "230.659", "L1": "230.408"}, "pinvreactive": {"L2": "269.257", "L3": "270.747", "L1": "281.205"}, "ts": {"val": "2022-04-30T07:37:59UTC"}, "ploadreactive": {"L2": "-185.329", "L3": "-67.524", "L1": "234.935"}, "state": {"val": "4101"}, "wloadprodq": {"L2": "3677347997", "L3": "5058670655", "L1": "3783842710"}, "pinv": {"L2": "-2915.945", "L3": "-2941.355", "L1": "-2929.194"}, "iextq": {"L2": "-15.215", "L3": "-18.135", "L1": "-15.136"}, "pext": {"L2": "-2484.383", "L3": "-2957.828", "L1": "-2466.003"}, "wextprodq": {"L2": "2197072304835", "L3": "2134969473089", "L1": "2756682738432"}, "wpv": {"val": "14004608404988"}, "winvconsq": {"L2": "58709692914", "L3": "37641033192", "L1": "122371452668"}, "pextreactive": {"L2": "83.929", "L3": "203.223", "L1": "516.140"}, "udc": {"neg": "-380.545", "pos": "381.070"}, "sext": {"val": "8015.244"}, "iextd": {"L2": "0.514", "L3": "1.246", "L1": "3.168"}, "ild": {"L2": "1.649", "L3": "1.660", "L1": "1.726"}, "pload": {"L2": "431.563", "L3": "-16.473", "L1": "463.190"}, "ilq": {"L2": "-17.858", "L3": "-18.034", "L1": "-17.979"}, "winvprodq": {"L2": "4387852532188", "L3": "4530315003509", "L1": "4384302101529"}, "il": {"L2": "12.691", "L3": "12.800", "L1": "12.769"}}' Traceback (most recent call last): File "/config/custom_components/ferroamp/sensor.py", line 137, in ehub_event_received update_sensor_from_event(event, ehub, store) File "/config/custom_components/ferroamp/sensor.py", line 131, in update_sensor_from_event sensor.add_event(event) File "/config/custom_components/ferroamp/sensor.py", line 623, in add_event self.process_events(now) File "/config/custom_components/ferroamp/sensor.py", line 630, in process_events if self.update_state_from_events(temp): File "/config/custom_components/ferroamp/sensor.py", line 752, in update_state_from_events pct = int(float(self.state) / 10) * 10 ValueError: could not convert string to float: 'unknown'
Exception in ehub_event_received when handling msg on 'extapi/data/ehub': '{"wloadconsq": {"L2": "22297563515019", "L3": "23128947679830", "L1": "15106501056370"}, "iloadd": {"L2": "5.837", "L3": "-5.036", "L1": "-0.079"}, "wextconsq": {"L2": "19975740691464", "L3": "20570665882684", "L1": "13471409102115"}, "ppv": {"val": "9191.379"}, "iext": {"L2": "8.307", "L3": "6.768", "L1": "12.509"}, "iloadq": {"L2": "8.910", "L3": "9.280", "L1": "0.528"}, "iace": {"L2": "0.000", "L3": "0.000", "L1": "0.000"}, "ul": {"L2": "229.878", "L3": "229.521", "L1": "230.824"}, "pinvreactive": {"L2": "268.205", "L3": "268.437", "L1": "285.304"}, "ts": {"val": "2022-04-30T07:38:25UTC"}, "ploadreactive": {"L2": "948.794", "L3": "-817.322", "L1": "-12.894"}, "state": {"val": "4101"}, "wloadprodq": {"L2": "3677347997", "L3": "5058952426", "L1": "3783842710"}, "pinv": {"L2": "-2905.713", "L3": "-2937.068", "L1": "-2948.029"}, "iextq": {"L2": "-8.966", "L3": "-8.817", "L1": "-17.534"}, " [... cut ...] mponents/ferroamp/sensor.py", line 131, in update_sensor_from_event sensor.add_event(event) File "/config/custom_components/ferroamp/sensor.py", line 623, in add_event self.process_events(now) File "/config/custom_components/ferroamp/sensor.py", line 630, in process_events if self.update_state_from_events(temp): File "/config/custom_components/ferroamp/sensor.py", line 752, in update_state_from_events pct = int(float(self.state) / 10) * 10 ValueError: could not convert string to float: 'unknown'

argoyle, 2022-05-01:
Hmmm. That did not look good. Are you running the latest version? Just so I look at the correct code from the stack trace.

bj00rn, 2022-05-02:
> Hmmm. That did not look good. Are you running the latest version? Just so I look at the correct code from the stack trace.

Yes, i believe so. 1.7.2

bj00rn, 2022-05-02:
Hmmm weird. after a restart of HA the issue now seems to be gone.

argoyle, 2022-05-02:
Seems like you had old values which were `unknown` which is very bad to to calculations on. 😂 Added a small check for that in #163

commit 96259067 (2022-05-02): fix: handle unknown value for percentage sensor

Closes #162
PR #163 (2022-05-04): fix: handle unknown value for percentage sensor
Closes #162
closed 2022-05-04 by argoyle