henricm/ha-ferroamp #265: Setup sanity check fails with "MQTT integration is not available"

angas, 2022-12-17:
As I'm new to Home Assistant I can't determine what's the main reason behind this, what's wrong/different with my configuration and why the sanity (introduced in 704919c682b86969b617515cdd255c8b93c64b7a) fails. Maybe the filtering by `mqtt.DOMAIN` is to optimistic?

A little research though shows that I get two entries during the setup from `async_entries`, the second (I think) is the one that is expected and not the one at index [0]. 

**First entry**
```
{"entry_id": "3207631bce4bed1e1bc8a5aed05053e1", "version": 1, "domain": "mqtt", "title": "MQTT", "data": {}, "options": {}, "pref_disable_new_entities": false, "pref_disable_polling": false, "source": "ignore", "unique_id": null, "disabled_by": null}
```

**Second entry**
```
{"entry_id": "9da04d74c231c5c8e1de355bb8d9a006", "version": 1, "domain": "mqtt", "title": "192.168.???.???", "data": {"broker": "192.168.???.???", "port": 1883, "username": "???", "password": "???", "keepalive": 60, "tls_insecure": true, "protocol": "3.1.1", "transport": "tcp", "discovery": true, "discovery_prefix": "homeassistant", "birth_message": {"topic": "homeassistant/status", "payload": "online", "qos": 0, "retain": false}, "will_message": {"topic": "homeassistant/status", "payload": "offline", "qos": 0, "retain": false}}, "options": {}, "pref_disable_new_entities": false, "pref_disable_polling": false, "source": "user", "unique_id": null, "disabled_by": null}
```

argoyle, 2022-12-19:
Hmmm, that sounds strange. Have you added the MQTT integration to your Home Assistant setup?

angas, 2022-12-19:
Yes, I have added the MQTT integration.

My setup is working fine if I change [this](https://github.com/henricm/ha-ferroamp/blob/master/custom_components/ferroamp/__init__.py#L59) line to use index 1 instead of 0.

In my `.storage/core.config_entries` file there is an (orphan) entry, corresponding to my index 0, with an empty data section that I think is the problem here. Unfortunately I can't tell how it got there.

Please let me know how you would like to proceed:

- I remove my first orphan entry and then forget about this.
- I create a pull request that handles my strange config situation.
- Any other solution you like...

argoyle, 2022-12-19:
Really strange. Do you have two config entries for MQTT in the UI (integrations-page) as well? If so, what happens if you remove the one which is not configured correctly? We can't default to 1 since most people won't have more than one config-entry I would say.

argoyle, 2022-12-19:
That code is a straight copy-paste from one of the built-in integrations from HA and was added by one of the MQTT integration maintainers so I would expect that to be correct.

closed 2022-12-19 by angas

argoyle, 2022-12-19:
Nice!

argoyle, 2022-12-19:
Could you create a PR with a relevant message-change?