henricm/ha-ferroamp #279: Problem after upgrading Ferroamp to version 5.2.2

umepali, 2023-01-11:
After Ferroamp support upgraded my system I’m not getting any updates on these values EnergyHub External Energy Consumed and EnergyHub External Energy Produced so the energy dashboard are not showing any consumption or production.

robinostlund, 2023-01-11:
@umepali can you install mqtt explorer and connect to your mqtt broker or directly to ferroamp and share your values?

umepali, 2023-01-11:
I have installed mqtt explorer and connected to ferroamp but how do i get any values?

robinostlund, 2023-01-11:
It should look like this, the values we need is in the extapi/ehub. So click on ehub and then on the copy to cliboard to the right of Value. Check example here: https://capture.dropbox.com/Zo3ialkhxUh6c97D

umepali, 2023-01-11:
{"gridfreq": {"val": "50.020"}, "soc": {"val": "80.000"}, "soh": {"val": "100.000"}, "ratedcap": {"val": "5760.000"}, "iace": {"L1": "0.000", "L2": "0.000", "L3": "0.000"}, "ul": {"L1": "233.877", "L2": "231.562", "L3": "233.111"}, "il": {"L1": "0.591", "L2": "0.596", "L3": "0.583"}, "ild": {"L1": "0.792", "L2": "0.798", "L3": "0.780"}, "ilq": {"L1": "0.070", "L2": "0.068", "L3": "0.072"}, "iext": {"L1": "6.031", "L2": "7.625", "L3": "4.585"}, "iextd": {"L1": "0.148", "L2": "0.243", "L3": "0.285"}, "iextq": {"L1": "8.175", "L2": "10.617", "L3": "6.342"}, "iloadd": {"L1": "-0.644", "L2": "-0.555", "L3": "-0.495"}, "iloadq": {"L1": "8.105", "L2": "10.549", "L3": "6.270"}, "iavblq": {"L1": "11.825", "L2": "9.383", "L3": "13.658"}, "iavblq_3p": {"val": "9.383"}, "iavbl": {"L1": "14.219", "L2": "12.492", "L3": "15.515"}, "sext": {"val": "4244.986"}, "pext": {"L1": "1351.949", "L2": "1738.418", "L3": "1045.380"}, "pextreactive": {"L1": "24.476", "L2": "39.789", "L3": "46.978"}, "pinv": {"L1": "11.576", "L2": "11.134", "L3": "11.868"}, "pinvreactive": {"L1": "130.978", "L2": "130.664", "L3": "128.571"}, "pload": {"L1": "1340.373", "L2": "1727.283", "L3": "1033.511"}, "ploadreactive": {"L1": "-106.502", "L2": "-90.875", "L3": "-81.593"}, "ppv": {"val": "0.464"}, "pbat": {"val": "0.000"}, "udc": {"pos": "380.177", "neg": "-379.790"}, "wextprodq": {"L1": "291678011", "L2": "71627038", "L3": "476130184"}, "wextconsq": {"L1": "944365973395", "L2": "1019024927665", "L3": "819766548967"}, "winvprodq": {"L1": "22885702543", "L2": "27034533042", "L3": "23849876769"}, "winvconsq": {"L1": "36963830391", "L2": "35101422352", "L3": "38179039469"}, "wloadprodq": {"L1": "264949679", "L2": "433739106", "L3": "84038877"}, "wloadconsq": {"L1": "1080830486034", "L2": "1175554379448", "L3": "936952468063"}, "wextprodq_3p": {"val": "565895729"}, "wextconsq_3p": {"val": "3237894807507"}, "winvprodq_3p": {"val": "79863067587"}, "winvconsq_3p": {"val": "124636285196"}, "wloadprodq_3p": {"val": "730411156"}, "wloadconsq_3p": {"val": "3193286105325"}, "wpv": {"val": "79844958534776"}, "wbatprod": {"val": "11325946273841"}, "wbatcons": {"val": "9050446384926"}, "state": {"val": "4101"}, "ts": {"val": "2023-01-11T14:46:01UTC"}}

robinostlund, 2023-01-11:
@umepali it seems correct, i got the same format and i am running on v5.0.2. Can you verify so that it haven't created new sensors in homeassistant? for example sensor.ferroamp_external_energy_produced -> sensor.ferroamp_external_energy_produced_2 or something?

I have requested a upgrade on my system to ferroamp so hopefully they upgrade it soon and then i can also see if i will get the same issues.

umepali, 2023-01-11:
No new sensors what I can see, I have 10 devices and 85 entities is that correct?
I can wait until you received the new version.

umepali, 2023-01-12:
That was the problem I have lost all Consumed and Produced numbers for 2022. Creating a support case with Ferroamp and hope they can get the numbers back.

argoyle, 2023-01-18:
I got my EnergyHub upgraded to 5.2.6 yesterday and all seems to work correctly. Closing this issue. If you still have problems we can re-open.

closed 2023-01-18 by argoyle