henricm/ha-ferroamp #283: Sensor Cost?

bj00rn, 2023-01-18:
What are these sensors?

[image]

sensor.energyhub_14_external_energy_consumed_cost
sensor.energyhub_14_external_energy_consumed_cost

Can't find any references in source for these.

robinostlund, 2023-01-18:
They are created by the energy integration in ha so not related to this integration 😊

bj00rn, 2023-01-18:
> They are created by the energy integration in ha so not related to this integration 😊

Ok I see, nothing to worry about then

closed 2023-01-18 by bj00rn

argoyle, 2023-01-18:
Love it when the community solves stuff before I even have seen it. 😆 ❤️

bj00rn, 2023-01-20:
> Love it when the community solves stuff before I even have seen it. 😆 ❤️

The naming of the sensor is a bit misleading. One gets the impression that it belongs to this integration.  But i guess that's out of our control 🤷

argoyle, 2023-01-20:
Yep, that's HA long term statistics/energy dashboard thingie doing it's magic.