henricm/ha-ferroamp #291: Battery entities not able to choose in Energy Dashboard

andreasropel, 2023-02-01:
Hi. 

I am not able to setup these two in energy dashboard. They don’t even show up as entities in the list when choosing battery consumption or production. Are they defined in correct way? 

```
Battery Energy Produced
Battery Energy Consumed
```

TIA 
—Andreas

argoyle, 2023-02-01:
Hmmm, that's strange. I have them in mine. Can you check the statistics-tav in Developer Tools and see if there is some problem with them perhaps?

andreasropel, 2023-02-01:
To me they looks ok! 
thanks for quick reply. 

[image]

argoyle, 2023-02-01:
That's the information the Energy-thingie uses

andreasropel, 2023-02-01:
You mean this? 

[image]

andreasropel, 2023-02-01:
I found the problem. Need to include them in recorder of course!! Thanks for guidance

argoyle, 2023-02-01:
No, the statistics-tab at the right side here 😄 
[image]

But good that you found the problem.

closed 2023-02-01 by argoyle