henricm/ha-ferroamp #292: Non-numeric value

argoyle, 2023-02-02:
Just noticed this in my log:
```
Sensor sensor.ferroamp_available_rms_current_for_load_balancing has device class current, state class measurement and unit A thus indicating it has a numeric value; however, it has the non-numeric value: unknown (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
```

robinostlund, 2023-03-02:
I also have this, do you think this is related to upgrade of the hub @argoyle ? i am on version 5.2.6

argoyle, 2023-03-02:
I thinks it's something new in HA since 2022.2. I'll see if I can solve it somehow when I have time.

argoyle, 2023-03-02:
As far as I have understood those three sensors needs to be configured in the hub before it's going to send values for them. Perhaps a better solution would be to not create those if no values comes from the hub?

robinostlund, 2023-03-03:
@argoyle agree, if not exists = skip or something :)

PR #308 (2023-03-18): feat: only add sensors for loadbalancing if present in message
Closes #292 @robinostlund
commit 6bac6eb6 (2023-03-18): feat: only add sensors for loadbalancing if present in message

Closes #292
closed 2023-03-18 by argoyle