henricm/ha-ferroamp #316: Possible memory leak

mickecamino, 2023-03-27:
I have installed Debian 11 server latest version, installed HA Core (latest) , HACS, MQTT and ha-ferroamp.
As soon as I enable ha-ferroamp and starts to receive data the server memory slowly decreases until HA crashes., rate is about 1 kb per minute.
To test this I have another instance of HA running with a lot of sensors, this server has been running flawless for about a year.
I installed and enabled ha-ferroamp on this server, as soon as I did this the memory slowly decreases.
There are two components involved, MQTT and ha-ferroamp. I am not sure where the memory is allocated and not released, it could be MQTT or it could be ha-ferroamp.
There are no errors in the logs (I have not enabled debug logs yet).
Versions used:
Home Assistant 2023.3.6 
Frontend 20230309.1 - latest 

HACS:
Integration version: | 1.31.0
Frontend version: | 20220906112053
Downloaded repositories: | 4

Debian GNU/Linux 11 (bullseye)

Python 3.10.10.

EnergyHub Extapi Version = 1.2.1

My python skills are limited, I am a PHP-coder,  I have not read up how variables are allocated in python or their sizes. 
It could be that my EnergyHub is manufactured in 2023 and has the latest firmware in it, Ferroamp could have changes something so that data is somewhat different than before.

I will enable debug logging for HA to see if I can get anything useful out of it.
I have the data from the memory sensor if needed.

argoyle, 2023-03-27:
That's not something I've seen or heard about before. I'll keep an eye on my memory-usage for a while now and please add any debug-info you might find.

argoyle, 2023-03-27:
My mem-usage has increased from 5.2Gb this morning to 5.6Gb right now. My Mosquitto broker has remained at 0.2% mem-usage since then so the problem is probably inside HA or the integration then. Have you changed the update interval of the integration in any way? The EnergyHub is a bit spammy sending a message a second at least. I've set mine to 10 seconds (default is 30 IIRC).

mickecamino, 2023-03-27:
The Update Interval is at default 30 seconds.
And yes, the EnergyHub is sending a lot of data:
extapi/data/ehub - transmission interval: 1s
extapi/data/eso - transmission interval: 5s
extapi/data/sso - transmission interval: 5s
extapi/data/esm - transmission interval: 60s
The latest API documentation, Revision E 2021-03-22, is here: https://ferroamp.com/wp-content/uploads/2023/01/Ferroamp-External-API-specification-E.pdf

I will install an older version of HA and see if it has the same problem and report the outcome.

mickecamino, 2023-03-27:
This is from my HA, you can clearly see the reboots that I've made. I am installing a brand new HA Core and I think I will start with 2022-01 version of HA and see if that release has the problem. If not, I will climb the release levels to see where the memory leak will occur. 
[image]

mickecamino, 2023-03-27:
Unfortunately HACS require 2022.11, so I had to start with that. Then ha-ferroamp requires 2022.12.0b0, so there is what I am right now. Will report back in a couple of hours.

mickecamino, 2023-03-27:
With HA version 2022.12.0b0, latest HACS and latest ha-ferroamp there is a memory leak. 
I installed a new Debian 11 (in Hyper-V 2019 server), installed Python 3.10.10 (from source), installed HA 2022.12.0b0 (via pip), installed HACS (via HACS install script for HA core), installed HA MQTT, then ha-ferroamp (via HACS) and from 5:50 PM to 9:00 PM memory consumption went from 1.3 GB to 1.090 GB.
Unfortunately I could not test with earlier versions of HA as HACS needed a fairly new version and ha-ferroamp wanted at least 2022.12.0b0.

mickecamino, 2023-03-27:
Hold a while. I *might* found the issue. I will implement it on my production server, the test server seems to hold the memory after I did some updating on Python. I'll be back

mickecamino, 2023-03-28:
Unfortunately, that did not help, still a memory leak. I am investigating further to see if I can find the leak.

mickecamino, 2023-03-28:
There is a memory leak in HASSIO as well with ha-ferroamp enabled, but I wonder in what module it is?
[image]
I will let the server run until it crashes to see if I can get anything out of it.

mickecamino, 2023-03-29:
Checked today. hassio seems to planing out with the memory, but it is swapping it to disk instead:
[image]
And the GUI is also showing this:
[image]
I will lhe hassio run until it crashes, then I will shut that down and trash it.
For my production server I will set up cron to restart HA every night to free up the memory.
Crude but effective :-)
I will log all MQTT for about 5 minutes to see if there is something in that data causing the memory leak.
The memory allocation seems to be in HA/MQTT not in ha-ferroamp, as far as I can understand the python code.

mickecamino, 2023-03-29:
Interesting, I checked hassio this afternoon and discovered this:
[image]
So I looked at the console and saw this:
[image]
hassio restarted ha after running out of memory.
As I have stated before, I am not a python coder so I don't know where to start looking for the leak. I have browsed ha-ferroamp code and I don't see anything there that do any memory allocation, so my best guess is that it is in the mqtt, ha or in a module used by python. I have not tested Python 3.11 (yet).
I have setup a trigger that restarts my HA every night at 2 AM, crude, but effective.
If you want me to do any more research let me know.

mickecamino, 2023-03-30:
Here is an odd thing. In hassio I disabled ha-ferroamp for a couple of minutes. I then enabled it, now memory is released back to the OS slowly. I will test this further and report back.

argoyle, 2023-03-30:
Really strange. As you say we don't really do any memory allocation in the integration so my guess is somewhere else. My machine is up to 8.9Gb mem-usage now. Having a box with 32Gb makes it a non-issue for quite some time of course and I will probably reboot it before it's a real problem as well.

mickecamino, 2023-03-31:
Looks like a bug in HA: https://github.com/home-assistant/core/issues/89885

argoyle, 2023-03-31:
Let's keep this open until it's solved.

bdraco, 2023-04-03:
The leak is likely here 
https://github.com/henricm/ha-ferroamp/blob/61b3f328fd1c5cf279c8480ef69a047ad5dec702/custom_components/ferroamp/sensor.py#L631

Nothing ever removes these AFAICT

argoyle, 2023-04-03:
Hmmm, I thought this line would take care of that: https://github.com/henricm/ha-ferroamp/blob/61b3f328fd1c5cf279c8480ef69a047ad5dec702/custom_components/ferroamp/sensor.py#L639

Far from a python expert though.

bdraco, 2023-04-03:
Is it possible the sensor is disabled and `self._added` is `False` so it gets the data but never calls that?

mickecamino, 2023-04-03:
It looks like Ferroamp is sending more data than defined in this module, at least these:
`
extapi/data/ehub:
"wextprodq_3p": {"val": "955998450236"}, 
"wextconsq_3p": {"val": "2120481960176"}, 
"winvprodq_3p": {"val": "1816349775315"}, 
"winvconsq_3p": {"val": "8498686944"}, 
"wloadprodq_3p": {"val": "141646443"}, 
"wloadconsq_3p": {"val": "2972476244754"}, 

And possibly:
"neg": "-379.749"}, 
`

Those are not in the latest API documentation.

argoyle, 2023-04-04:
> That line should take care of it, but it doesn't seem like its being called. It should cause it to get garbage collected next cycle. `.clear()` might be a better option though to avoid recreating the object
> 
> Is it possible the sensor is disabled and `self._added` is `False` so it gets the data but never calls that?

You might be on to something there. At least for some users there are entities that are not added. Will look into it. Thanks!

argoyle, 2023-04-04:
> It looks like Ferroamp is sending more data than defined in this module, at least these: ` extapi/data/ehub: "wextprodq_3p": {"val": "955998450236"}, "wextconsq_3p": {"val": "2120481960176"}, [...]

Those are for reactive thingies. Not sure actually. I have not enabled those for myself.

mickecamino, 2023-04-04:
I have made the change now and will check if there are any leaks in an hour or so.

henricm, 2023-04-04:
Wow - good find! Thanks a lot @mickecamino and @bdraco for the in depth analysis!

mickecamino, 2023-04-05:
Unfortunately it did not fix my issue. Still got memory leak. I will debug  a little more to see where the leak is.

argoyle, 2023-04-05:
Hmmm, you got data for the reactive sensors though, correct? Those are the ones which were handled in #322

mickecamino, 2023-04-05:
Yes, however, I have battery enabled in the portal, but it is not installed yet.
I have run profiler and I can see that there are dicts leaking, but I can't see where they are coming from.
I have disabled Nordpool to see if it is that module affects anything.

mickecamino, 2023-04-05:
I started up my testserver, updated ferroamp via hacs, and will let it run for a while to see if it is fixed running this server. 
This server have only HACS and Ferroamp installed. I will report back probably tomorrow.

mickecamino, 2023-04-06:
I have upgraded HA Core to 2023.4.0, and will let it run for a couple of hours with memory logging. Fingers crossed

mickecamino, 2023-04-07:
This is really strange. On my production server I have HA 2023.4, Python 3.10.10 running on Debian 11 and the following modules:
HA-modules: MQTT, Profiler (for debugging), SMHI, Sun and Forecast.Solar
HACS with latest nordpool and latest Ferroamp Sensors

This server leaks memory by a couple of kilobytes per minutes.
[image]

I also have a testserver with:
HA 2023.4, Python 3.10.10 running Debian 11 and the following modules:
HA-modules: MQTT, Home Meteorologisk institutt (Met.no), Profiler and Sun
HACS with Ferroamp Sensors
This leaks some memory, but nowhere near the production server.

[image]

I am thinking of building a new server, install bare modules and copy the database over so that I don't lose any history and see if that solves the issue.

I will continue to investigate

bdraco, 2023-04-07:
Some memory use increase is normal for the first few hours as caches build. If it doesn't level off you have a leak

mickecamino, 2023-04-07:
Same setting, 30 seconds. I have now compared all Python-libs in the env against the test server and upgraded a few in my production server and restarted HA.

mickecamino, 2023-04-07:
I think I have nailed it. 
I stopped HA, deleted all files in the /home/homeassistant/.homeassistant/custom_components/ferroamp/__pycache__ folder. 
Started HA and it seems to be stable now. I will check it tomorrow (Swedish time) and report the final.
I will add back all previous integrations one by one to see if anything there affects the memory.

mickecamino, 2023-04-09:
Sigh, it's back to eating memory. 
I added Norpdool and Solar Forecast and it started to eat memory.
I then removed Nordpol and Solar forecast, it sill eats memory. I cleaned the cache, restarted server etc. 
Next thing to try is to set up a new server with only HA, HACS and Ferroamp.
This wont happen until next week

mickecamino, 2023-04-11:
I have requested that Ferroamp upgrade my Energyhub from 5.2.1 to 5.3.2. I don't know if this will solve the issue, but I need 5.3.2 for my CTEK charger

danielolsson100, 2023-04-15:
This memory leakage is stable now with the latest version and I think therefore that this issue can now be closed.
[image]

mickecamino, 2023-04-15:
I still have a memory leak, so lets keep this open until:
1. Our battery is installed (will be installed on Wednesday)
2. EnergyHub is updated to 5.3.2, I am running 5.2.1 at the moment, ticket created at Ferroamp requesting a firmware update.

mickecamino, 2023-04-19:
After installation of the PSM-battery the memory leak seems to have disappeared :-)
I will update and reboot the production server tonight and will report tomorrow if it's still stable.

argoyle, 2023-04-19:
Ok. Thanks for the information. Will need to think a bit regarding the sensors in the Energy Hub and why any of them would "store" events without processing them.

argoyle, 2023-04-19:
The ESM/ESO sensors are only created once we receive messages originating in those components so the problem must be in the EnergyHub-sensors.

mickecamino, 2023-04-19:
Here is the strange part: I have a testserver that has been stable after your first fix. I have compared databases, configuration files etc, but has not been able (yet) to spot any differences that affected the memory leak. I have tweaked a couple of differences between then but not been able to spot the difference.
So It might be that the initial sensors received differs from what the EH sends after an upgrade?
I am at 5.2.1 on my EH, but have requested an upgrade to 5.3.2, not yet received.

mickecamino, 2023-04-20:
After reboot and adding some more integrations I am pleased to say that there is no more memory leak. It is rock solid.
Here is a log where HA complains about six sensors, I guess that it is these that caused the memory leak as the battery was enabled in the energyhub but it was not installed:
sensor.ferroamp_battery_energy_produced
sensor.ferroamp_battery_energy_consumed
sensor.ferroamp_system_state_of_charge
sensor.ferroamp_system_state_of_health
sensor.ferroamp_battery_power
sensor.ferroamp_total_rated_capacity_of_all_batteries

```
2023-04-19 09:45:42.782 WARNING (MainThread) [homeassistant.components.sensor] Sensor sensor.ferroamp_battery_energy_produced has device class energy, state class total_increasing and unit kWh thus indicating it has a numeric value; however, it has the non-numeric value: unknown (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
2023-04-19 09:45:42.804 WARNING (MainThread) [homeassistant.components.sensor] Sensor sensor.ferroamp_battery_energy_consumed has device class energy, state class total_increasing and unit kWh thus indicating it has a numeric value; however, it has the non-numeric value: unknown (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
2023-04-19 09:45:42.807 WARNING (MainThread) [homeassistant.components.sensor] Sensor sensor.ferroamp_system_state_of_charge has device class battery, state class measurement and unit % thus indicating it has a numeric value; however, it has the non-numeric value: unknown (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
2023-04-19 09:45:42.810 WARNING (MainThread) [homeassistant.components.sensor] Sensor sensor.ferroamp_system_state_of_health has device class None, state class measurement and unit % thus indicating it has a numeric value; however, it has the non-numeric value: unknown (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
2023-04-19 09:45:42.812 WARNING (MainThread) [homeassistant.components.sensor] Sensor sensor.ferroamp_battery_power has device class power, state class measurement and unit W thus indicating it has a numeric value; however, it has the non-numeric value: unknown (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
2023-04-19 09:45:42.815 WARNING (MainThread) [homeassistant.components.sensor] Sensor sensor.ferroamp_total_rated_capacity_of_all_batteries has device class None, state class None and unit Wh thus indicating it has a numeric value; however, it has the non-numeric value: unknown (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
```

argoyle, 2023-04-20:
Sounds likely. Will try to get some unit-test going for that. Thanks for the pointers. ❤️  Let's keep this issue open until I have been able to investigate some more since this will likely bite someone else in the future.

PR #322 (2023-04-05): fix: don't store events if sensor is not present
Might fix #316. Could you try this change locally @mickecamino?
commit a7420628 (2023-04-05): fix: don't store events if sensor is not present

Might fix #316
PR #334 (2023-04-24): fix: only create battery specific sensors if present in message from …
…EnergyHub

Fixes #316

@mickecamino @danielolsson100
closed 2023-04-24 by argoyle