henricm/ha-ferroamp #317: Sensors not working

Monewalker, 2023-03-29:
Hi 

Since 28/3 my sensors are not updating. I sent in a ticket to Ferroamp and they enabled MQTT again as I thought that they might have disabled it during some update. But it is still not working. Do you know of any issues? 
See attached pictures for examples.

Best Regards

Monewalker 

[image]
[image]

argoyle, 2023-03-30:
Can you check if you receive any messages on the MQTT-broker? Easiest done by going to the integrations-page, click on Configure on the MQTT integration, enter `extapi/data/ehub` in the listen to a topic-field and press start listening. You should get about 1 message per second if correctly configured.

Monewalker, 2023-03-30:
Hi

I dont seem to get any response from extapi/data/ehub.
But if i login directly to Ferroamp hub using mqtt explorer i see all data.
If i login to homeassistant via mqtt explorer i dont see extapi/data/ehub.

Mqtt has been working for a long time but at 12:20 28/3 the data disappeared. No change in my config and if i check the config file for mqtt communication to ferroamp hub it looks okay.

Any ideas ?

Best regards

Monewalker

argoyle, 2023-03-30:
I would probably look into the MQTT broker then. Are you using a bridge connection? Have you tried restarting the MQTT broker?

Monewalker, 2023-03-30:
Hi

I went to mqtt broker addon and enabled debugging. After that i checked the sensors and they where showing values again.
I disabled debugging and it still works.

I have restarted homeassistant several times with no result. So Im confused why this worked.

Thank you for your assistant.
The ticket can be closed.

Best Regards

Monewalker

Mvh Michael

argoyle, 2023-03-30:
The broken probably restarted when you enabled debug-logging. Addons are not restarted when you restart HA core (unless you reboot the whole machine it's running on of course).

Glad it got resolved.

closed 2023-03-30 by argoyle