henricm/ha-ferroamp #318: Issue with v1.10.2

u1malmberg, 2023-04-02:
I installed latest version two days ago and the fix for v1.10.2 ([743cb44](https://github.com/henricm/ha-ferroamp/commit/743cb440ed280e5adda2f699f2f2c78d6b819785))  is causing issues with the statistics. 

For two days the daily production utility "daily" meter have been completely off.  Today showing ~5MWh instead of 40-50kWh. 

Checking on the data I can see both days I have had one value reported from Ferroamp that have been off by 18% and 22%, both happening within 15mins around 1400hrs both days.  

Attached image shows total energy produced reported around 27MWh, then there is one erroneous at 22MWh, from there on the daily is then at 4.7MWh rather than the ~45kWh.

[image]

If this fix is to handle an actual reset, why not block down to 90% rather than the current 10% (also it appears to be actually less than 10%).  A reset is 100% after all! :) 

So, the assumption that wrong data only is 10% less is not accurate.

robinostlund, 2023-04-02:
I also noticed this yesterday when i restarted by hub 😊

argoyle, 2023-04-02:
Hmmm, maybe we should revert that change @henricm ? What versions do you guys have on your EnergyHubs? Just trying to find some kind of pattern here on why/when this is happening.

u1malmberg, 2023-04-02:
I am running 5.0.3 if that makes a difference.  Not sure where these erroneous values comes from as it is just one in the serie, then back to correct values again.  There used to be similar issues before it was set to "total_Increasing", but that was fixed more than a year ago and it has worked like a charm ever since.

It would not be so bad if it did not mess up the daily, monthly, yearly aggregates.

henricm, 2023-04-02:
How does it continue? Will EH continue to report 22 MWh and increasing from there or does it go back to 27 MWh?

Could you post more data and possibly also how this looks in energy cloud?
When sensors are affected?

Home assistant uses 10% margin when trying to determine if there has been a reset or if there are error values for total-increasing sensors, so that's the reason for using the same. 
Also, even if an actual reset happens it may take time before a system is back up again. And the counters may have moved during this time so I don't think we can use eg 90%. 

If we should change this back it would be good to know more details when this happens and the nature of the incorrect values. Rather than a % maybe we may also need to use a time frame?

argoyle, 2023-04-02:
Maybe we could ignore outliers somehow?

u1malmberg, 2023-04-02:
As mentioned earlier, not sure why it occurred, never actually seen it before, although that does not mean much as you seldom look at the actual data -- row by row.  But, at least in these two instances, it was just one value that stood out, before and after the series is correct.  

[image]

henricm, 2023-04-02:
Would be good to know how "wrong" the data could be. I guess ferroamp has no clue about this issue?

henricm, 2023-04-02:
> Maybe we could ignore outliers somehow?

Is it doable to determine this scenario where a value is off a few seconds - and ignore it in that case? We do some average calculation already for values entering in between the configured fetch interval, right? Maybe it can bee hooked in there somehow?

u1malmberg, 2023-04-02:
hmm, I cannot see any traces of this in the reports in FA portal -- although they are on a fairly high level.  

Just looked at the dataset from 1st of jan this year up until today, exported it into csv and these two instances from today and yesterday are the only entries that are "off".  well.. based on sorting all the rows and checking from lowest ... just these two.  

Strange, I mean what are the odds!?  only two errors this year are since after I updated the integration.  I did update from 1.9-ish!? not sure exact "from" version (is it possible to check in HA somewhere!?), could it be one of the other patches that somehow could mess up the data? 

Also, last two days have been the sunniest this year :) , so I guess we cannot rule out that the errors from FA could be more frequent under heavy load or something.

henricm, 2023-04-02:
We ignored all lower values so I guess that's why you don't see any of this before updating to latest.

henricm, 2023-04-02:
A quick fix for this particular case would be to change the 10 % limit to 25 % instead and hope that's about how wrong the data can get.
I don't think I've seen this issue except in cases where there has been some issue with my EH. I recently got the SD card replaced for example, resetting some (not all) counters.

robinostlund, 2023-04-02:
I am running 5.3.2. I will doublecheck tomorrow but i think it was showing 0 at one point after reboot of the hub

robinostlund, 2023-04-02:
[image]

argoyle, 2023-04-02:
> > Maybe we could ignore outliers somehow?
> 
> Is it doable to determine this scenario where a value is off a few seconds - and ignore it in that case? We do some average calculation already for values entering in between the configured fetch interval, right? Maybe it can bee hooked in there somehow?

I'm thinking we could have a sliding average of the last minute or so and compare with that?

u1malmberg, 2023-04-02:
The big question is if to leave it also for tomorrow to see if I get the same results around 1400 again! 😃

henricm, 2023-04-02:
> [image]

Thanks! Have you reported this to ferroamp? Doesn't look healthy.

robinostlund, 2023-04-03:
@henricm haven't reported this to ferroamp, will do that now :)

robinostlund, 2023-04-03:
Created a ticket to ferroamp now. Noticed that these issues only occur on data from SSO + ESO, not the hub itself.
[image]
[image]

henricm, 2023-04-03:
> I'm thinking we could have a sliding average of the last minute or so and compare with that?

Maybe not work that well with the scenario @robinostlund is experiencing where values dropping to 0. I'm thinking that maybe this is happening to @u1malmberg too, but the average-calculation we do in between updates maybe hides this fact?

One idea could be to to max() of the x latest values. Then it will take a few updates before an actual reset is taken into account, but a few error values won't affect the sensor.

robinostlund, 2023-04-03:
or should we just add a if >= 0 skip or something? As it seems to go to zero when it is rebooted? 😄

argoyle, 2023-04-03:
> Maybe not work that well with the scenario @robinostlund is experiencing where values dropping to 0. I'm thinking that maybe this is happening to @u1malmberg too, but the average-calculation we do [...]

Not sure how that would be handled with regards to average values etc. I think I'm more inclined to do the skip 0-values as @robinostlund suggested. That is if the actual low value seen by @u1malmberg was 0 (it was averaged out in the graphs of course and not sure how to know what the actual value was).

u1malmberg, 2023-04-03:
Was going to tell you that I did not receive any wrong data today around 1400 (as usual), but then it happened around 16:30-ish instead.     About the same "distance", ~20%, away from the correct values.  

@argoyle, I am not seeing 0 values, just 20%-ish off values. 

The 25% limit discussed above might work for me, but then again, since we do not know why the data is wrong, then we do not really know what the diff could be in the future... 

[image]

henricm, 2023-04-03:
> Not sure how that would be handled with regards to average values etc. I think I'm more inclined to do the skip 0-values as @robinostlund suggested. That is if the actual low value seen by [...]

Maybe worth a try - if it's always 0-values that's causing this issue. Regardless this needs to be done in the average calc right? Since it's already too late if the incorrect values have been included in the average sum?

Otherwise I was thinking we could keep a list of the latest 10 or so values. Just as we keep values in between the configured update interval to do the average calc. And use Max instead of average for total-increasing energy counters which should make more sense than average calc. That could then also work if other types of junk values can appear.

But it will of course complicate things a bit 🙂.

henricm, 2023-04-03:
> Was going to tell you that I did not receive any wrong data today around 1400 (as usual), but then it happened around 16:30-ish instead.     About the same "distance", ~20%, away from the correct [...]

You do not see all values that EH sends since the sensor in HA is only updated every 30 second but values from EH arrives more often than that. The values that arrives in between are collected and used in average calculation every 30 second (or the interval you've configured). To see all values maybe you could connect an mqtt explorer or if it's possible to enable more logging of mqtt messages in HA?

robinostlund, 2023-04-03:
@henricm what version are you running? they did some changes regarding the measurement in latest version: https://support.ferroamp.com/en/support/solutions/articles/47001241579-software-release-5-3-2

henricm, 2023-04-04:
@robinostlund I'm on 5.0.3

henricm, 2023-04-06:
Closing. Please reopen or file new issue if the problem comes back.

closed 2023-04-06 by henricm