henricm/ha-ferroamp #356: Control maximum export from HA

LudvigD, 2023-06-11:
I don't know if it's okay to make a feature request as an issue so please remove it if not.

Currently we have a new problem with negative electric prices when the sunshine as best.
I have contacted Ferroamp but now their support email is closed.

They have a feature on their portal to limit the export of electricity, but i don't know if it's available to mqtt. Anyhow i will keep trying to contact them, as the new electricity prices makes it necessary to limit the export of solar energy.

I suggest that is this repository could include that option.

I will add info here if I receive an answer from Ferroamp. 

Kind regards
Ludvig Derning

argoyle, 2023-06-12:
That functionality is not available over MQTT unfortunately.

danielolsson100, 2023-07-08:
@LudvigD This can be achieved by utilizing the ferroamp portal and send http API requests but this is not supported and outside this integration since it's not MQTT based.
Look at https://gist.github.com/danielolsson100/1c99bae4e2605540854a73b1e13a6309 and se concept code for this.
I am using it myself to control Import / Export values.

argoyle, 2023-08-21:
Closing this for now since there is no possibility to do this right now.

closed 2023-08-21 by argoyle