henricm/ha-ferroamp #364: EnergyHub DC Link Voltage state is zero

bj00rn, 2023-07-07:
Seems that the sensor state sums up to zero.

[image]

Is this really the correct way to represent DC voltage? I have no domain knowledge here whatsoever here, it just appears to me that the current state value seems useless.

Nominal DC link voltage is 760V (720-780V) according to specs, which leaves 380+380 within that range..

[image]

From the looks of it would seem that the correct way of summing up dc link voltage would be something like self._attr_native_value = int(abs(neg) / count + pos / count) 

offending line:
https://github.com/henricm/ha-ferroamp/blob/56df5e9f542d0b4d9f843acd52ab5107cdcfd614/custom_components/ferroamp/sensor.py#L759

argoyle, 2023-07-07:
No domain knowledge at my end either but I see how you're thinking. @danielolsson100 or @mickecamino perhaps knows more?

danielolsson100, 2023-07-07:
@argoyle I have no clue in this matter since I am not using this value at all, I don't even store the history data in the DB for this sensor ;)

jonasbkarlsson, 2023-07-07:
It should be `pos / count - neg / count`.

Also, it shouldn't be `int()`. Would be better to use something like `round(float(pos / count - neg / count), 2)`.

mickecamino, 2023-07-07:
The value from Ferroamp is a floa, so it should be a float in the code:
[image]

argoyle, 2023-07-07:
> It should be `pos / count - neg / count`.
> 
> Also, it shouldn't be `int()`. Would be better to use something like `round(float(pos / count - neg / count), 2)`.

That would result in ~760 in the case @bj00rn had, right? Can someone provide a PR with those changes?

mickecamino, 2023-07-07:
Hmmm, it seems that I looked at the wrong sensor, Here is the correct one:
[image]

bj00rn, 2023-07-08:
> > It should be `pos / count - neg / count`.
> > Also, it shouldn't be `int()`. Would be better to use something like `round(float(pos / count - neg / count), 2)`.
> 
> That would result in ~760 in the case @bj00rn had, right? Can someone provide a PR with those changes?

I can give it a shot

closed 2023-07-08 by argoyle

bj00rn, 2023-07-09:
Works on my machine (TM) :-)

[image]

PR #365 (2023-07-08): fix: correct calculation dc link voltage sensor state
Correct calculation of dc link voltage sensor state. Seems the values where reversed

fixes #364
commit a4a64172 (2023-07-08): fix: correct calculation dc link voltage sensor state

Correct calculation of dc link voltage sensor state. Seems the values where reversed

fixes #364