henricm/ha-ferroamp #405: Battery sensor spamming logs

bj00rn, 2023-09-12:
Battery sensors seems to be spamming logs like crazy. Sensor is unavailable (no battery installed).

Not sure why or when this behaviour started

```
2023-09-12 10:33:06.608 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_consumed value None seems to be zero. Ignoring
2023-09-12 10:33:07.566 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_produced value None seems to be zero. Ignoring
2023-09-12 10:33:07.567 INFO (MainThread) [custom_components.ferroamp.sensor] 
2023-09-12 10:33:08.576 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_produced value None seems to be zero. Ignoring
2023-09-12 10:33:08.576 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_consumed value None seems to be zero. Ignoring
2023-09-12 10:33:09.567 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_produced value None seems to be zero. Ignoring
2023-09-12 10:33:09.567 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_consumed value None seems to be zero. Ignoring
2023-09-12 10:33:10.567 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_produced value None seems to be zero. Ignoring
2023-09-12 10:33:10.567 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_consumed value None seems to be zero. Ignoring
2023-09-12 10:33:11.547 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_produced value None seems to be zero. Ignoring
2023-09-12 10:33:11.547 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_consumed value None seems to be zero. Ignoring
2023-09-12 10:33:12.568 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_produced value None seems to be zero. Ignoring
2023-09-12 10:33:12.568 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_consumed value None seems to be zero. Ignoring
2023-09-12 10:33:13.555 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_produced value None seems to be zero. Ignoring
2023-09-12 10:33:13.555 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_consumed value None seems to be zero. Ignoring
2023-09-12 10:33:14.557 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_produced value None seems to be zero. Ignoring
2023-09-12 10:33:14.557 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_consumed value None seems to be zero. Ignoring
2023-09-12 10:33:15.566 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_produced value None seems to be zero. Ignoring
2023-09-12 10:33:15.567 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_consumed value None seems to be zero. Ignoring
2023-09-12 10:33:16.557 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_produced value None seems to be zero. Ignoring
2023-09-12 10:33:16.557 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_consumed value None seems to be zero. Ignoring
2023-09-12 10:33:17.549 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_produced value None seems to be zero. Ignoring
2023-09-12 10:33:17.549 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_consumed value None seems to be zero. Ignoring
2023-09-12 10:33:18.559 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_produced value None seems to be zero. Ignoring
2023-09-12 10:33:18.561 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_consumed value None seems to be zero. Ignoring
2023-09-12 10:33:19.547 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_produced value None seems to be zero. Ignoring
2023-09-12 10:33:19.547 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_consumed value None seems to be zero. Ignoring
2023-09-12 10:33:20.565 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_produced value None seems to be zero. Ignoring
2023-09-12 10:33:20.565 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_consumed value None seems to be zero. Ignoring
2023-09-12 10:33:21.575 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_produced value None seems to be zero. Ignoring
2023-09-12 10:33:21.575 INFO (MainThread) [custom_components.ferroamp.sensor] sensor.energyhub_14_battery_energy_consumed value None seems to be zero. Ignoring
```

argoyle, 2023-09-12:
Hmmm, strange. Aren't those sensors that are only added if they get a value? Or was that other sensors?

bj00rn, 2023-09-12:
> Hmmm, strange. Aren't those sensors that are only added if they get a value? Or was that other sensors?

Yup
https://github.com/henricm/ha-ferroamp/blob/2af4429aad43dfc20e4e1d10bc45ed61056ec47c/custom_components/ferroamp/sensor.py#L2580

bj00rn, 2023-09-12:
Here's a payload, sensor data does not seem present.

When whas that check introduced? I can see the sensors under integration

[image]

[image]

```json
{"gridfreq": {"val": "50.010"}, "iace": {"L1": "0.000", "L2": "0.000", "L3": "0.000"}, "ul": {"L1": "234.132", "L2": "234.959", "L3": "235.665"}, "il": {"L1": "4.910", "L2": "4.857", "L3": "4.899"}, "ild": {"L1": "1.100", "L2": "1.090", "L3": "1.087"}, "ilq": {"L1": "-6.859", "L2": "-6.784", "L3": "-6.841"}, "iext": {"L1": "4.329", "L2": "3.832", "L3": "4.498"}, "iextd": {"L1": "1.532", "L2": "2.205", "L3": "1.566"}, "iextq": {"L1": "-5.813", "L2": "-4.737", "L3": "-6.000"}, "iloadd": {"L1": "0.432", "L2": "1.115", "L3": "0.479"}, "iloadq": {"L1": "1.046", "L2": "2.047", "L3": "0.841"}, "iavblq": {"L1": "25.813", "L2": "24.737", "L3": "26.000"}, "iavblq_3p": {"val": "24.737"}, "iavbl": {"L1": "24.081", "L2": "23.289", "L3": "24.212"}, "sext": {"val": "2973.941"}, "pext": {"L1": "-962.379", "L2": "-787.010", "L3": "-999.842"}, "pextreactive": {"L1": "253.632", "L2": "366.341", "L3": "260.959"}, "pinv": {"L1": "-1135.551", "L2": "-1127.101", "L3": "-1139.986"}, "pinvreactive": {"L1": "182.112", "L2": "181.094", "L3": "181.138"}, "pload": {"L1": "173.172", "L2": "340.091", "L3": "140.145"}, "ploadreactive": {"L1": "71.520", "L2": "185.247", "L3": "79.821"}, "ppv": {"val": "3505.104"}, "udc": {"pos": "380.577", "neg": "-380.077"}, "wextprodq": {"L1": "25090105566800", "L2": "22845710593184", "L3": "21039603455884"}, "wextconsq": {"L1": "29751813679488", "L2": "42942721916386", "L3": "53119945116989"}, "winvprodq": {"L1": "30957724510500", "L2": "30954630063839", "L3": "32469446152052"}, "winvconsq": {"L1": "757024129041", "L2": "408593975029", "L3": "113079174211"}, "wloadprodq": {"L1": "9913253457", "L2": "6581174446", "L3": "17793438741"}, "wloadconsq": {"L1": "36164943544750", "L2": "52476626161394", "L3": "67044298933022"}, "wextprodq_3p": {"val": "46696989742511"}, "wextconsq_3p": {"val": "70887820874566"}, "winvprodq_3p": {"val": "62980260805466"}, "winvconsq_3p": {"val": "206082042929"}, "wloadprodq_3p": {"val": "4879309331"}, "wloadconsq_3p": {"val": "86969889203923"}, "wpv": {"val": "102014858598424"}, "state": {"val": "4101"}, "ts": {"val": "2023-09-12T09:47:41UTC"}}
```

bj00rn, 2023-09-12:
ah ok, introduced in https://github.com/henricm/ha-ferroamp/commit/167fbcd1adb5bbdc82f35b4ef30f9fdc828b37f5.

I guess my sensors where already created before then and the fix would be deleting them manually right?

bj00rn, 2023-09-12:
> Yes, I think that should solve your problem.

A bit unfortunate with the log frequency though, might be worth mentioning somewhere.

I shot my self in the foot with my (docker default) log configuration and HA docker logs ate my disk :-)

bj00rn, 2023-09-12:
Removing sensors manually stopped the spamming. Had to restart HA for it to take effect

bj00rn, 2023-09-12:
Edit: logs are back, even after manually deleting sensors and restarting HA.  reopening

[image]

bj00rn, 2023-09-12:
Tried debugging after deleting the sensors. Seems that sensor is still present sensors, with it's state being unknown

[image]

bj00rn, 2023-10-31:
@argoyle any pointers here? No battery sensors, and yet the log message pops up on every data refresh.

[image]

argoyle, 2023-10-31:
Sorry, haven't had the time to investigate yet. Will try to squeeze it in soon.

bj00rn, 2023-11-14:
There is a throttle class we can use until we get this sorted proper. I can do a PR

from homeassistant.util import Throttle
[…]
And then right before the update method:
@Throttle(MIN_TIME_BETWEEN_UPDATES)
def log_missing_sensor_data()
    [...]

bj00rn, 2024-01-15:
@argoyle did some more research on this. It seems FerroampSensor is missing `async_will_remove_from_hass` callback to delete the entity from the corresponding `store`. Or Am I missing something here?

```python
    async def async_added_to_hass(self) -> None:
        """Handle entity which will be added."""
        await super().async_added_to_hass()
        state = await self.async_get_last_state()
        if state and state.state != "unknown":
            self._attr_native_value = state.state
        self.hass.data[DOMAIN][DATA_DEVICES][self.config_id][self.device_id][
            self.unique_id
        ] = self
        self._added = True

    async def async_will_remove_from_hass(self) -> None:
        del self.hass.data[DOMAIN][DATA_DEVICES][self.config_id][self.device_id][
            self.unique_id
        ] # <- delete sensor
        await super().async_will_remove_from_hass()
```

argoyle, 2024-01-15:
You might be totally correct on that point. Could you add it and see if it works better? It's difficult for me to test since I have a battery.

bj00rn, 2024-01-16:
> You might be totally correct on that point. Could you add it and see if it works better? It's difficult for me to test since I have a battery.

Oops, i accidentally deleted my previous comment. Sensor registration/store and check_presence logic is a bit messy and should probably be cleaned up IMHO, this issue hardly warrants that kind of effort though.

I made a PR to suppress the log messsage for `EnergyFerroampSensor` taking check_presence into account instead.

PR #479 (2024-01-17): fix(sensor): do not log warning if sensor is not present
- [x] do not log warnings in energy sensor when sensor data is not present in event

If sensor data is missing in event, `EnergyFerroampSensor` will write a warning to log to on every update, even if `check_presence=True`

The reason for this behaviour is that being that the `check_presence` check is done in the base class.

The issue is fixed by adding an additional check in `EnergyFerroampSensor`.

This fix applies to all `EnergyFerroampSensor` with `check_presence=True`
 - `Battery Energy Produced`
 - `Battery Energy Consumed`

fixes  [... cut]
closed 2024-01-17 by argoyle