henricm/ha-ferroamp #411: Sensors in energy wrong

bluboyo, 2023-09-28:
Hi!

Ive tried to add the ferroamp integration in the energy tab of HA. But all the sensors are wrong. For example total solar is the total solar produced day. What you want is solar power. But i cant choose that. The readings is completely of the board. Compering with the FA cloud. 

Jus to be clear. The sensors are correct outside the energy part. But inside you cant chose the correct ones.

argoyle, 2023-09-29:
I think you misunderstand what the energy dashboard is using. It's an "energy" dashboard and not a "power" dashboard so the sensors that you can choose from is correct. See the [Energy Dashboard](https://github.com/henricm/ha-ferroamp#energy-dashboard) for more information on how to configure it correctly.

bluboyo, 2023-09-29:
If you look at the realtime dashboard you can see that the readings is wrong.
I get what you are doing but the realtime is wrong. You need to add the realtime sensors and let HA sample the data. You are adding the totals and that screws everything up.

Can you add the other sensors so I at least can use it as intended?

bluboyo, 2023-09-29:
Can you add
sensor.ferroamp_consumption_power
sensor.ferroamp_grid_power
sensor.ferroamp_solar_power

Thats the sensors needed to get the realtime working

robinostlund, 2023-09-29:
Those sensors already exists @bluboyo ?
I am using those for my realtime energy card.

bluboyo, 2023-09-29:
Yes but you cant choose them in the HA energy part. If you make your own sure. But the point is to use the energy tab and have it all there.

regards

Den 29 sep. 2023 08:53, robinostlund skrev:

robinostlund, 2023-09-29:
No it isn't possible to use them for ha energy part as the requirement there is to use total increasing values, so you should use these sensors defined in: https://github.com/henricm/ha-ferroamp#energy-dashboard

Power is newer going to be total increasing

bluboyo, 2023-09-29:
I understand that. But if you look at the realtime dashboard you can see that the readings are total. Not realtime. The animation is moving. Thats means realtime.

So realtime data and let HA be the database. Then then statistics will come with time.

regards

-------- Originalmeddelande --------
Den 29 sep. 2023 09:05, robinostlund skrev:

robinostlund, 2023-09-29:
HA energy dashboard isn't realtime, it updates the values every hour based from the total counters.
Are you using another realtime dashboard integration?

bluboyo, 2023-09-29:
there is a realtime part of the HA dashboard. Looks like the ferroamp.

robinostlund, 2023-09-29:
Hmm, i think we need more information, where is the realtime part documented from HA? I haven't seen any support for realtime from Homeassistant, only by using custom components.

bluboyo, 2023-09-29:
[image]
[image]
[image]
Look at these screenshots. Should look like the FA dashboard. but its says im using 20,2 kwh when its 2.1

I havent looked at the documentation. This is just logic.

robinostlund, 2023-09-29:
Ahaa yes, but what you are searching for isn't supported by HA. HA energy dasbhoard is using kWh to count the usage and is not providing live data. What you are searching for is my previous example by using the power flow card which will show you a dashboard like in Ferroamp portal :) HA Energy Dashboard = kWh, not Power (W).

bluboyo, 2023-09-29:
Then that dashboard from HA is pointless. Should be removed and just show statistics. Il check out your dashboard :)

opsiorobinostlund, 2023-09-29:
HA dasbhoard is for showing your daily usage so you can keep track of your usage and cost, as kWh is energy and kW is power. 2 different things :). So go to HACS and install power flow card and provide my settings, then you will have what you are searching for :)

bluboyo, 2023-09-29:
Thank you. Worked like a charm. But HA should still remove that animation. Per day isnt really what you expect form a moving animation.

robinostlund, 2023-09-29:
Well don't really agree, i like it, but feel free to bring that up with the HA team :)

bluboyo, 2023-09-29:
Hehe. Anyway thx for the integration!

argoyle, 2023-09-29:
Closing this now that all is sorted. 😊

closed 2023-09-29 by argoyle