henricm/ha-ferroamp #417: Hi and thanks for the very good follow-up :), MQTT had to be turned on in the local ferroamp from the supplier. Everything now seems to work.

Moogels, 2023-10-05:
Hi and thanks for the very good follow-up :), MQTT had to be turned on in the local ferroamp from the supplier. Everything now seems to work.

_Originally posted by @bopbrev in https://github.com/henricm/ha-ferroamp/issues/72#issuecomment-916988244_

Moogels, 2023-10-05:
Hi, who did you contact? Is it possible to do it yourself?

argoyle, 2023-10-05:
You need to contact Ferroamp support here: https://support.ferroamp.com/sv-SE/support/tickets/new

Moogels, 2023-10-10:
I have contact ferroamp support, and I think that it's working when I look at the log from MQTT explorer? 
But not in homeassistant  : /
Could you please check my screen shots for any faulty settings? :  

[image]
[image]
[image]
[image]
[image]

argoyle, 2023-10-10:
Hmmm, it looks like your custom config is not being used. My Mosquitto logs says this on restart:
[image]

My config is in `share/mosquitto/energyhub.conf` and looks like this:
```
connection energyhub
address 192.168.1.97
remote_username extapi
remote_password superSecretPassword
topic extapi/# both
```

Moogels, 2023-10-11:
It was somehow solved by turning off NGIX proxy manager add on :) Thx!!

argoyle, 2023-10-11:
Nice! Closing this then.

closed 2023-10-11 by argoyle