henricm/ha-ferroamp #483: no sensors visible

bjornwiberg, 2024-01-25:
ive added to mqtt client and this through hacs
tested to listen and the messages in the mqtt client and they are received correctly from extapi/data/ehub

but theres not extra sensors populated, only the one for checking if an update is available or not

heres the diagnostics:

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.1.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Stockholm",
    "os_name": "Linux",
    "os_version": "6.1.63-haos-raspi",
    "supervisor": "2023.12.1",
    "host_os": "Home Assistant OS 11.4",
    "docker_version": "24.0.7",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "nordpool": {
      "version": "0.0.14",
      "requirements": [
        "nordpool>=0.2",
        "backoff"
      ]
    },
    "ferroamp": {
      "version": "1.14.1",
      "requirements": []
    },
    "hacs": {
      "version": "1.34.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "mqtt",
    "name": "MQTT",
    "codeowners": [
      "@emontnemery",
      "@jbouwh"
    ],
    "config_flow": true,
    "dependencies": [
      "file_upload",
      "http"
    ],
    "documentation": "https://www.home-assistant.io/integrations/mqtt",
    "iot_class": "local_push",
    "quality_scale": "gold",
    "requirements": [
      "paho-mqtt==1.6.1"
    ],
    "is_built_in": true
  },
  "data": {
    "connected": true,
    "mqtt_config": {
      "broker": "192.168.68.60",
      "port": 1883,
      "username": "**REDACTED**",
      "password": "**REDACTED**",
      "keepalive": 60,
      "tls_insecure": false,
      "protocol": "5",
      "transport": "tcp",
      "discovery": true,
      "discovery_prefix": "homeassistant",
      "birth_message": {
        "topic": "homeassistant/status",
        "payload": "online",
        "qos": 0,
        "retain": false
      },
      "will_message": {
        "topic": "homeassistant/status",
        "payload": "offline",
        "qos": 0,
        "retain": false
      }
    },
    "devices": [],
    "mqtt_debug_info": {
      "entities": [],
      "triggers": []
    }
  }
}
```

argoyle, 2024-01-25:
You also have to add the integration itself under Settings / Integrations (Inställningar / Enheter och tjänster and then the Integrationer-tab). Press the add integration button and search for Ferroamp.

bjornwiberg, 2024-01-25:
thanks, that did it, i forgot to add it as a integration after i've downloaded it

closed 2024-01-25 by bjornwiberg