henricm/ha-ferroamp #535: Question: Incorrect statistics when MQTT was down over new year

bj00rn, 2024-04-24:
Had my MQTT bridge down over Christmas and when I started it again (on 9th of January) wrong values seem to have been recorded in statistics.

I got negative statistic for export/import corresponding to last years total production/consumption.

I guess the fix is to correct the statistic manually. It would be interesting to know why this happened in the first place.

Is this expected statistics behaviour when a sensor has been offline?

[image]

bj00rn, 2024-04-24:
Edit: I resolved the problem and removed all the incorrect negative statistics for all `total increasing` sensors.

argoyle, 2024-04-25:
Hmmm, it shouldn't happen just because the sensor goes unavailable I think. Those are strictly increasing sensors if I don't misremember and those are reset if the values decrease. Could be that you were bitten by the problem some others have had that the EnergyHub decreased some sensor values in some update.

bj00rn, 2024-04-25:
> Hmmm, it shouldn't happen just because the sensor goes unavailable I think. Those are strictly increasing sensors if I don't misremember and those are reset if the values decrease. 

> Could be that you were bitten by the problem some others have had that the EnergyHub decreased some sensor values in some update.

Hard to tell since state (kept for two weeks) has been cleared from state tables and moved to statistics by now. When I removed the statistics outliers the statistic recovered fine though. I highly suspect this is related to MQTT bridge coming back online. I will try checking what the EH is reporting.

brandgul01, 2024-05-14:
I might have a similar problem. installed MQTT bridge via mosquito yesterday and had a gaps weeks with no statistics saved. Before I connected directly with HAs client to Energyhubs broker. Today my energy-view is bananas. I found a large minus stat in mars with the method you used bjOOrn and removed it. but it didn't solve the problem. If I look at the sensors individually everything looks ok to me but added to HA internal energy page it doesn't make sense at all. I will try to continue to solve this, with the limited HA knowledge I have. 
[image]
[image]
[image]

argoyle, 2024-05-15:
The energy dashboard uses the long term statistics (which of course is based on the sensor values). It stores differences over a given time period for the sensor values. Some of the sensors are of the "strictly increasing" type and when a value is received from the EH that is lower than the previous value the sensor is reset which will cause a spike. The value should be strictly increasing from EH as well but I know that there was a problem for some users in a firmware update version some time back.

bj00rn, 2024-05-15:
@brandgul01 adjust all weird looking statistic entries to `0` fixed it for me. You probably forgot some. Once you have found the exact time the process is straightforward

brandgul01, 2024-05-15:
@bj00rn I used the button "find extreme values". Very useful. Will go through all again. Also strange that I get no values on the home-icon. But I guess it's caused by the same fault.
@argoyle Thanks for your input as well. If the problem is on the EH-side with a suddenly changed value to the negative I guess it's impossible for anyone then Ferroamp to adjust that. Should be possible to see in the web portal stats though.

brandgul01, 2024-05-15:
Update: Found an extra extreme value, changed it to "0" and now everything works.