henricm/ha-ferroamp #555: EnergyHub Control Status when active on FCR

nsimb, 2024-05-25:
Feature request

Hi and thanks for all the work with this.
Just wanted to check if it's possible to update the control status when FCR markets.
Or keep the response in someplace so automations can know that it's no idea to try to change the configuration.

Background, I Am currently testing out FCR market with a retailer/balance with ferroamp to use the battery on FCR market. When that happens ferroamp will disable the MQTT commands and take over the control. When the activation is ended on FCR they will release the battery back to MQTT handling with the status it had when they did take control. this is how it is supposed to work at least according to the support.

Since there is no way today to know when they are active and not it ends up that my automation tries to act and it's no point
Tested a bit during an activation of what was returned, and it's a nak.
request:

{
  "transId": "04029caa-1abb-11ef-b292-94c691186b60",
  "cmd": {
    "name": "discharge",
    "arg": 0
  }
}

response:
{
  "status": "nak",
  "transId": "..............................",
  "msg": "failed to execute command: \"ext-api-set (): Ongoing\" taint present, which does not allow setting [\"ext-api-set: Ongoing\"] (All non-tolerations are: [\"ext-api-set: Ongoing\",\"battery-maintenance-regular: Ongoing\",\"battery-maintenance-critical: Ongoing\",\"fcr-event: Ongoing\"])"
}

I have not yet found anything other in the mqtt packages so far that that is usable sadly

Let me know if I can help out in any way

argoyle, 2024-05-25:
I actually have exactly the same problem when sending charge and discharge commands after being upgraded to 5.7.2 and I'm not even in any FCR project. 😳 I'll try to get in touch with Ferroamp regarding this and see if there is anything we might see and do via MQTT. Or has @danielolsson100 already been in contact with them regarding this perhaps?

danielolsson100, 2024-05-26:
I sent an RFC to Ferroamp regarding a FCR/remote control flag in the MQTT data so we can know when the battery is managed by Ferroamp cloud in any way.
But there has been no more respons than "we will send it to the developers".

nsimb, 2024-05-26:
Hade the same discussion with the support on an earlier firmware 5.5.0-FCR.1, they fixed my problem in next beta release. Did then bring up that it would be good with state in integration when they take control. Will reach out to them again and ask if something is in the roadmap. Did hear that more retail company’s is on the way into FCR with Ferroamp so the demand for the development should increase.

nsimb, 2024-11-11:
anyone noticed changes how it behaves during active on FCR? 
I hade a long mail conversation with the retailer that then was in contact with Ferroamp. My main complaint was that if my system was in Discharge(0)/charge(0) and Ferroamp did take over the control (blocking part of API) they did set it to "auto" even when there was no activativation for FCR. Just standby
This was draining my battery o really cheep prices. 

I notioced this weekend that the battery was active and API was blocked. however the battery did not act as it was in Auto. It kept the  inactive stage until activation was needed.

Anyone ells have noticed similar behaver? fingers crossed that they working on the FCR and API logic with a bigger change

danielolsson100, 2024-11-11:
Perhaps the answer on your question is in one of the registers in Modbus TCP at https://github.com/henricm/ha-ferroamp/issues/652 
I have not looked into that yet, so I am just guessing.
Some people that work at FA seems to be "pointing/recommending" to use Modbus TCP for controlling the battery.

nsimb, 2024-11-11:
> Perhaps the answer on your question is in one of the registers in Modbus TCP at #652 I have not looked into that yet, so I am just guessing. Some people that work at FA seems to be "pointing/recommending" to use Modbus TCP for controlling the battery.

Did spend some time to look into the document, would say its the possibility to add grid limitations that is the difference. Allso (without exact comparing) get a feeling there is less data published compare to the MQTT.
Seams to bundle all battery data and the ESO. And treating it as one.
Could be some errors in the doc, since its strange that they talk about Power and then use kWh ( Battery Power Reference Input 36107 float [-inf, inf) kWh -ve power implies charging)

Sadly I dont see anything related the topic of FCR, status or similar

we can proabebly close this feture request, and we see when if ferroamp releases somthing new

danielolsson100, 2025-01-18:
I'll again open a case again regarding this matter now to FA to see if there is any status flag when the battery is controlled from the cloud.
I haven't found it in the Modbus TCP data but the data might be undocumented.