henricm/ha-ferroamp #652: MODBUS?

bj00rn, 2024-11-03:
Since release 5.8 modbus api modbus api supports export/import limits and battery control. 

Maybe implement support? Or is it better to have as a separate integration?

https://support.ferroamp.com/en/support/solutions/articles/47001265164-software-release-5-8

https://support.ferroamp.com/helpdesk/attachments/47307477251

argoyle, 2024-11-03:
For me this feels like a separate integration. There isn't really anything in common between the ways of communication apart from the target device being the EH. I don't have final say on this though. @henricm, what's your stance?

robinostlund, 2024-11-03:
I think there is just a matter of time until these new functions will arrive to mqtt

nsimb, 2024-11-03:
Was hoping there was something in there regarding FCR market, but did not find anything.
Yes hopefully limitation functions comes to Mqtt api as well.
Fingers crossed that they working on a bigger update.

henricm, 2024-11-03:
Agree it might be better with another integration. However, if there are functionality that ferroamp won't make available in mqtt it might make sense to implement it here as well.
Does modbus require ferroamp to enable it as in the mqtt case or is it enabled by default? 
Also, how far would one get, using the basic [support for modbus](https://www.home-assistant.io/integrations/modbus/) available in Home Assistant, to control export import limits?

bj00rn, 2024-11-03:
> Agree it might be better with another integration. However, if there are functionality that ferroamp won't make available in mqtt it might make sense to implement it here as well. Does modbus [...]

looks like it needs to be enabled :-(

[image]

robinostlund, 2024-11-03:
It needs to be enabled as with mqtt.
I have started to map the different sensors: https://pastebin.com/HZtBPZ2c

danielolsson100, 2024-12-15:
Hi  @robinostlund 
I tested your Modbus TCP sensors and the most of them seems to work but not the "Import Threshold" because the portal flips out when I give it at try.
I am guessing that its related to the data type that is defined as float according to the spec at https://ferroamp.com/wp-content/uploads/2023/04/Ferroamp-Modbus-TCP-Specification-RevD-2.pdf
Do you have a working examle for it that you can share?

note. I think that the corresponding "export values" should be negative input data.

```
    - name: Ferroamp Import Threshold
      state: "{{ states('sensor.ferroamp_import_threshold_system_value') }}"
      availability: "{{ is_number(states('sensor.ferroamp_import_threshold_system_value')) }}"
      set_value:
        # set import threshold
        - action: modbus.write_register
          data_template:
            hub: ferroamp_modbus
            address: 8002
            slave: 1
            # value: "{{ value }}"
            value: 
              - 0
              - "{{ '%x' % (value) | pack('>f') | unpack('>hh') }}"
 
        # apply import settings
        - action: modbus.write_register
          data_template:
            hub: ferroamp_modbus
            address: 8006
            slave: 1
            value:
              - 1

danielolsson100, 2024-12-28:
The answer to my question was to understand the float structure in Modbus TCP in IEEE 754
Thanx for sharing @robinostlund 

In my case with 16A house fuse / 11kW Grid load as Max
Import Threshold should be between -11000 to 0 
> Example: Use the battery if the grid load is lower than -11kW to save the fuse.

Export Threshold should be between 0 to 11000
> Example: PV production: put the grid overload in the battery to save the fuse

More cases can apply:
> Example: If the hour price is high so it's more logical to sell the energy than dumping it in the battery but I haven't build that logic yet since the is no "high energy sun mode" right now in December. 

```    - name: FMB Export Threshold
      state: "{{ states('sensor.fmb_export_threshold_system_value') }}"
      availability: "{{ is_number(states('sensor.fmb_export_threshold_system_value')) }}"
      unique_id: fmb_export_threshold
      set_value:
        # set import threshold
        - action: modbus.write_register
          data_template:
            hub: ferroamp_modbus
            address: 8012
            slave: 1
            value: >
              {% set value = value | float(0) %}
              {% set packed_value = pack(value, ">f") %}
              {% set high_word = unpack(packed_value, ">H", offset=2) | abs %}
              {% set low_word = unpack(packed_value, ">H") | abs %}
              [ {{ high_word }}, {{ low_word }} ]

        # apply import settings
        - action: modbus.write_register
          data_template:
            hub: ferroamp_modbus
            address: 8016
            slave: 1
            value:
              - 1
      step: 1
      min: -11000
      max: 0

argoyle, 2025-11-23:
Sweet! I still see this as a totally separate integration though since the only common factor is that it talks to the same hardware.

zgnusrx, 2025-11-24:
@danielolsson100 Yep got it to work, thanks!

I'm using what you are suggesting. Modbus, only alter import/export thresholds. Discharge? Both to -13k. Charge? +13k. Self consumption? 0. Battery pause/peak shaving: 0/7000. It works like a charm. Have a script that sets thresholds once tomorrow_valid turns true and ensures battery stays within 93-11%. Feel bad for people using FerroAI.

argoyle, 2026-04-02:
Totally reasonable! Thanks for creating it!

closed 2026-04-02 by argoyle